stm32-rs/stm32-rs #431: Licensing problem

VP-, 2020-09-09:
As mentioned in #152, the STM32 SVDs are proprietary. In particular, the license requires anybody who uses them to restrict the usage of their own software to only allow it to run on genuine ST microcontrollers. The generated crates are, except for the patched bits, automatically derived from and completely include the proprietary SVDs. As I understand it, this means the usage restrictions (and all other terms of the ST license) still apply to the generated crates. How can you then license them under pure Apache and MIT licenses without any additional conditions?

If my understanding is correct and if you decide to fix the license of the generated crates instead of re-creating all the SVDs from scratch: Since the usage restriction violate freedom 0 of the Free Software Definition, can you recommend any alternatives for use in free software?

adamgreig, 2020-09-09:
I can't provide legal advice to you, if you are concerned then you should obtain relevant advice yourself. I don't intend to change the licence on this repository, and couldn't anyway since I don't own the copyright to the many contributions.

The generated crates don't include any SVD files, neither original nor modified.

I'm not aware of any other projects which provide register definitions for STM32s which are not based on the ST SVD files or C header files, and if they existed, they'd presumably be based on the ST reference manuals instead, which you might or might not consider materially different.

closed 2020-09-09 by adamgreig

VP-, 2020-09-10:
> ... change the licence on this repository, and couldn't anyway since I don't own the copyright to the many contributions.

That last part is not true. According to the [last section of the README](https://github.com/stm32-rs/stm32-rs#contribution), contributions are licensed under MIT and Apache licenses. Both are permissive licenses, i.e. allow anybody to re-distribute a derivative work of the contributions under a different license. That different license only needs to include the conditions of the original license you pick (either Apache or MIT or both).

The last point also applies to the license of the SVDs. You can add the restrictions from the Apache or MIT license, but you cannot remove the restrictions imposed by the original ST license.

> The generated crates don't include any SVD files, neither original nor modified.

That's like saying "The compiled executable doesn't include any source files." The important part is the fact that the automatically generated crates are a derivative work of the SVDs and the ST license therefore still applies. My point about "completely include" was only meant to highlight the fact that we can't even claim fair use, since the translations of the entire SVDs are re-distributed.

> I'm not aware of any other projects which provide register definitions for STM32s which are not based on the ST SVD files or C header files, and if they existed, they'd presumably be based on the ST reference manuals instead, which you might or might not consider materially different.

Good point about the difference. The material difference is which parts of manuals or header files are included in the distributed software. If it's only the names, addresses, and types of registers and their bit fields, then it's only a collection of facts about the microcontroller, and is not even copyrightable. If OTOH snippets of manuals, the comments from C headers, or any corresponding descriptions from SVD files are included, then the result is a derivative work of the work of the author of the manuals/comments/descriptions. Depending on how much is copied, fair use may apply, but certainly not when copying the entire comments/descriptions from a file.

I guess my questions boil down to: Does svd2rust include all descriptions from the SVD files in its output, and if yes, do the original SVD files even contain any descriptions beyond names? (Since I potentially face a clean-room re-creation of the crates, I didn't download the zip file myself yet.)