stm32-rs/stm32-rs #455: STM32L4R9AII6 support?

shymega, 2020-11-04:
Hi,

I'm looking to start work on a kernel for a STM32L4, specifically, a STM32L4R9AII6, and I can see there doesn't _appear_ to be support for that chip yet (unless I'm wrong?), and I was wondering if there's anything I can do is get it merged into mainline?

Thanks!

adamgreig, 2020-11-05:
You're right that we don't currently support the L4R9, but we _do_ already have an SVD file for it, so the minimum to add support is to create a new YAML file in `devices/` which points to `stm32l4r9.svd`, and add the part to `stm32_part_table.yaml` too.

It's likely the SVD can already be described by a number of patches in `peripherals/` and might want some fixes from `devices/common_patches/`, you could compare to some similar L4+ parts and also use the `scripts/matchperipherals.py` script to see if any apply cleanly.

For an example of a recent PR that added some new devices, check out https://github.com/stm32-rs/stm32-rs/pull/395.

I'd be very happy to merge a PR adding the L4R9 (and other L4R if you wanted!), but I doubt I'll have time to do it myself soon.

shymega, 2020-11-05:
hey,

thanks for getting back to me. 

w/ regards to the YAML file - that'd just be a top-level key, `svd`? I think it needs some more keys though.

thanks for the tip about the patches, and the PR. will take a look at the script...

if i need help with it, can i ask anything here? cheers.

adamgreig, 2020-11-07:
Just the `_svd` top-level key would be enough to generate a crate, all the other keys are to modify the SVD file in some way (fix errors, add descriptions for field variants, etc). Most of those modifications are in shared files so ones that apply to other L4 chips might be directly applicable, and can be included in the new stm32l4r9 file.

Sure, you can ask here or open a PR with your work-in-progress and ask there, either is fine.

shymega, 2020-11-07:
I'll ask here for now, working locally before a PR.

I'm not quite sure how to name the device, and the other L4R STM32 chips - I can see there's a convention for chips like `STM32L4x1. Now, the STM32L4R-series of STM32 chips doesn't exactly have that naming conventions -- the website identifies those chips as "STM32L4+ Series", so I was wondering what you'd advise for naming?

Other than that, I think I have the bare minimum done for the STM32L4R9AII6. Problem right now is that I don't _have_ the chip. That should be sorted soon...

ButtNaked, 2021-04-06:
Is anyone working on this?

Have some work done specifically on L4R9, want to contribute.

shymega, 2021-04-06:
I haven't worked on it for a while. Do you have a repo & branch for your work? I haven't been able to obtain a chip yet...

ButtNaked, 2021-04-07:
No, working locally atm. Though managed to make some basic hal peripheral to work.
Original synopsys-otg-usb driver works okay too.

shymega, 2021-04-07:
Just saw your PR. Looking good...

adamgreig, 2021-04-08:
Closed by #532, thanks @ButtNaked!

closed 2021-04-08 by adamgreig