stm32-rs/stm32-rs #457: STM32F0 I2C TXE register is marked read-only in SVD, but may be written to

barafael, 2020-11-26:
The datasheet RM0091 page 682 says "This bit can be written to ‘1’ by software in order to flush the transmit data register I2C_TXDR". I am not sure if this is wrong in the svd, or if this is just a very uncommon operation warranting an unsafe block and direct register manipulation.

barafael, 2020-11-27:
To clarify, I think it is actually necessary to set that bit during I2C configuration. 

See here: https://github.com/barafael/stm32f0-rtic-smbus-slave/blob/main/src/main.rs#L190

Without setting that bit, somehow the TXDR messes up resulting in a wrong transmission.

wallacejohn, 2021-03-29:
This issue may be confusing the TXE field in the SPI SR register (stm32f0x2.svd line 3077 mentioned above) with the TXE field in the I2C ISR register (stm32f0x2.svd line 4460). The SPI TXE field is read-only as described above and also matches the SPI chapter in RM0091. The I2C TXE field is read-write in the SVD and PAC and also matches the I2C chapter in RM0091.

closed 2022-05-31 by barafael