stm32-rs/stm32-rs #483: Conflicting cortex_m versions (take 2)

little-arhat, 2021-01-24:
```
warning: Linking globals named 'CORE_PERIPHERALS': symbol multiply defined!

error: failed to load bc of "cortex_m-5c0c4694ee41d0fc.cortex_m.63gufsr0-cgu.0.rcgu.o":

error: aborting due to previous error; 1 warning emitted
```

cortex-m 0.6, loaded via this stm32f3 conflicts with cortex-m 0.7 used by every other crate. Is there a way to fix that?

```
├── alt-stm32f30x-hal v0.31.0
│   ├── bitrate v0.1.1
│   ├── bobbin-bits v0.1.1
│   ├── cast v0.2.3
│   │   [build-dependencies]
│   │   └── rustc_version v0.2.3
│   │       └── semver v0.9.0
│   │           └── semver-parser v0.7.0
│   ├── cortex-m v0.7.0
│   │   ├── bare-metal v0.2.5
│   │   │   [build-dependencies]
│   │   │   └── rustc_version v0.2.3 (*)
│   │   ├── bitfield v0.13.2
│   │   ├── embedded-hal v0.2.4
│   │   │   ├── nb v0.1.3
│   │   │   │   └── nb v1.0.0
│   │   │   └── void v1.0.2
│   │   └── volatile-register v0.2.0
│   │       └── vcell v0.1.3
│   ├── embedded-hal v0.2.4 (*)
│   ├── nb v1.0.0
│   ├── stm32f3 v0.12.1
│   │   ├── bare-metal v0.2.5 (*)
│   │   ├── cortex-m v0.6.0
│   │   │   ├── aligned v0.3.4
│   │   │   │   └── as-slice v0.1.4
│   │   │   │       ├── generic-array v0.12.3
│   │   │   │       │   └── typenum v1.12.0
│   │   │   │       ├── generic-array v0.13.2
│   │   │   │       │   └── typenum v1.12.0
│   │   │   │       ├── generic-array v0.14.4
│   │   │   │       │   └── typenum v1.12.0
│   │   │   │       │   [build-dependencies]
│   │   │   │       │   └── version_check v0.9.2
│   │   │   │       └── stable_deref_trait v1.2.0
│   │   │   ├── bare-metal v0.2.5 (*)
│   │   │   └── volatile-register v0.2.0 (*)
│   │   ├── cortex-m-rt v0.6.13
│   │   │   ├── cortex-m-rt-macros v0.1.8 (proc-macro)
│   │   │   │   ├── proc-macro2 v1.0.24
│   │   │   │   │   └── unicode-xid v0.2.1
│   │   │   │   ├── quote v1.0.8
│   │   │   │   │   └── proc-macro2 v1.0.24 (*)
│   │   │   │   └── syn v1.0.59
│   │   │   │       ├── proc-macro2 v1.0.24 (*)
│   │   │   │       ├── quote v1.0.8 (*)
│   │   │   │       └── unicode-xid v0.2.1
│   │   │   └── r0 v0.2.2
│   │   └── vcell v0.1.3
│   └── void v1.0.2
├── asm-delay v0.9.0
│   ├── bitrate v0.1.1
│   ├── cortex-m v0.7.0 (*)
│   └── embedded-hal v0.2.4 (*)
├── cortex-m v0.7.0 (*)
├── cortex-m-rt v0.6.13 (*)
├── embedded-hal v0.2.4 (*)
├── nb v0.1.3 (*)
├── panic-abort v0.3.2
└── stm32f3 v0.12.1 (*)
```

(seems that similar issues was raised before: https://github.com/stm32-rs/stm32-rs/issues/184)

adamgreig, 2021-01-24:
Thanks for reporting, and for the cargo tree output. This will become fixed when cortex-m 0.6.5 is released (currently 0.6.5-alpha is out), which should be very soon. I'm really surprised it's pulling in 0.6.0 in your case; I've usually seen it go for 0.5.8 or something else that doesn't have the `links` field set in Cargo.toml.

In any event this is out of the hands of stm32-rs; it's entirely down to cortex-m having to ensure it's only linked _once_, but occasionally needing semver-incompatible releases. The next release of stm32-rs will support cortex-m >0.5.8,<0.8, I expect, but the cortex-m 0.6.5 release should fix this problem before that.

adamgreig, 2021-02-28:
@little-arhat this should now be fixed by the latest releases of cortex-m (and also stm32-rs), is it resolved for you?

little-arhat, 2021-03-07:
@adamgreig fixed indeed! Thank you!

closed 2021-03-07 by little-arhat