stm32-rs/stm32-rs #501: STM32H753 has 3 bits of wwdg prescaler while only 2 are in the pac

diondokter, 2021-02-19:
Datasheet of the H742/H743/H753/H750:
[image]

Config: https://github.com/stm32-rs/stm32-rs/blob/master/peripherals/wwdg/wwdg.yaml

I don't know how this crate works, so if somebody can instruct me with the rough steps, I could do it.

adamgreig, 2021-02-19:
Thanks for the report! There are two parts to the fix: first patching the SVD to increase the field width from 2 to 3 bits, and second adding extra enumeration variants. For the first, something like (untested):

```yaml
WWDG:
  CFG:
    _modify:
      WDGTB:
        bitWidth: 3
```

This could go into a new file like `devices/common_patches/h7_wwdg.yaml`, and be added to any H7 that has this problem, or you could just add it to `devices/stm32h753.yaml` to fix it just on this chip (if it doesn't affect other H7s).

The second part is slightly more annoying. The simplest option is probably to make an extra file in `peripherals/wwdg/wwdg_3bit_wdgtb.yaml` that contains something like:

```yaml
WWDG:
    CFG:
        WDGTB:
            _replace_enum:
                Div1: [0, "Counter clock (PCLK div 4096) div 1"]
                Div2: [1, "Counter clock (PCLK div 4096) div 2"]
                Div4: [2, "Counter clock (PCLK div 4096) div 4"]
                Div8: [3, "Counter clock (PCLK div 4096) div 8"]
                Div16: [4, "Counter clock (PCLK div 4096) div 16"]
                <etc>
```

and then include it anywhere that also includes the larger register.

diondokter, 2021-02-19:
I got some help from yatekii on matrix. He proposed creating a `peripherals/wwdg/h7_wwdg.yaml` just like the g4 series one that extends the normal peripheral. This keeps them separate which might be nice if any every need any more patching.

What do you think?

adamgreig, 2021-02-19:
Sure, that works fine too, you could exactly copy the `g4_wwdg.yaml` since it seems to do what you want. I suspect that actually the G4 and H7 use identical WWDG and you could instead rename it `wwdg_v2.yaml` and include it in both places, which might be nicer. We can always split it up again if they need to diverge.

commit c1c971d7 (2021-02-19): Merge #502

502: WWDG prescaler fix for H7 r=adamgreig a=diondokter

Fixes #501 

- Renamed the G4 wwdg to wwdg_v2
- Created patch for the bit width
- Added the patch to the H7 devices (I checked the datasheets and they all have the bigger prescaler)
PR #502 (2021-02-19): WWDG prescaler fix for H7
Fixes #501 

- Renamed the G4 wwdg to wwdg_v2
- Created patch for the bit width
- Added the patch to the H7 devices (I checked the datasheets and they all have the bigger prescaler)
closed 2021-02-19 by bors[bot]