stm32-rs/stm32-rs #503: Request for STM32WLE5 svd update

ofauchon, 2021-02-22:
Hi all, 

STM32WLE5 svd was updated on 18 Jan 2021 on st.com website: 
https://www.st.com/en/microcontrollers-microprocessors/stm32wlex.html#cad-resources

Here is the version provided on st.com : 

```
   <name>STM32WLE5_CM4</name>
   <version>1.4</version>
   <description>STM32WLE5_CM4</description>  
   <cpu>
    <name>CM4</name>
    <revision>r0p1</revision>
```

And the current version provided by stm32-rs:

```
   <name>STM32WLE5_CM4</name>
   <version>1.0</version>
   <description>STM32WLE5_CM4</description>  
   <cpu>
    <name>CM4</name>
    <revision>r0p1</revision>
```

What is the process to have it updated ? 

Thanks
Olivier

adamgreig, 2021-02-28:
Hi @ofauchon, thanks for the report. If you have time, you could open a PR that replaces the file `svd/vendor/en.stm32wl.zip` with the updated version. Since it looks like the `devices/stm32wle5.yaml` file doesn't contain any changes yet anyway, there should be no problem swapping to the new version.

ofauchon, 2021-03-02:
Hi @adamgreig, 

I made PR#507 for stm32wl series SVDs update. 
It seems the new bundle provides more .svd files compared to the current one: 

  * Current vendor/en.stm32wl.zip : 
stm32wle5.svd

  * New vendor/stm32wl_svd.zip
stm32wl5x_cm0p.svd
stm32wl5x_cm4.svd
stm32wle5_cm4.svd

I guess the new bundle now contains descriptor for both STM32W5x (dual core CM0/CM4) and STM32LE5 (single core CM4)

Hope that helps. 

Olivier

adamgreig, 2021-03-15:
Closed by #507, thanks for the PR!

closed 2021-03-15 by adamgreig

newAM, 2021-04-03:
For the next person who finds this issue: The new SVDs do contain some important things, such as `rcc.cr.msirgsel` which is not present in the old SVDs (which the current `0.13.0` release is based off of).

I made a temporary repo here with the generated rust as a temporary solution until the next release: https://github.com/newAM/stm32wl-rs

e.g.
```toml
[dependencies.stm32wl]
git = "https://github.com/newAM/stm32wl-rs.git"
rev = "031c2ecc3664c8ef3f642fbce075b2c238d5ff1b"
features = ["stm32wl5x_cm4"]
```

Thanks for updating this @ofauchon !

adamgreig, 2021-04-04:
Just so you know, the latest build is always available at https://github.com/stm32-rs/stm32-rs-nightlies which can also be used as a git dependency in Cargo (but avoid specifying a revision, as it wipes out the history every time there's a merge to master in this repo).

newAM, 2021-04-04:
> Just so you know, the latest build is always available at https://github.com/stm32-rs/stm32-rs-nightlies 

Whoops, I definitely did not know that, thank you!