stm32-rs/stm32-rs #504: STM32U5

jonathanpallant, 2021-02-25:
Cortex-M33, ultra lower power device. New out today. Let's get to it!

https://www.st.com/en/microcontrollers-microprocessors/stm32u5-series.html

jonathanpallant, 2021-02-25:
Ah, main availability won't be until September and it looks like CubeMX (and hence the packs containing the SVD files) won't be around until then.

adamgreig, 2021-09-29:
Update from September: the GitHub repo exists and has C headers, but no sign of an SVD file there or on st.com yet.

milesgranger, 2022-01-14:
Seems like they heard you talking about it; [available here](https://www.st.com/en/microcontrollers-microprocessors/stm32u5-series.html#cad-resources), uploaded on the same day. :sweat_smile:

jerabaul29, 2022-02-28:
Just being curious: what do you think is a typical / reasonable timeline for a dedicated crate for this MCU? :) .

jonathanpallant, 2022-02-28:
I have no need for the chip now, so I won't be writing anything (unless I'm paid to do so).

jerabaul29, 2022-08-07:
The board is indeed in stock at Mouser for example at the time of writing this :) : https://no.mouser.com/ProductDetail/STMicroelectronics/NUCLEO-U575ZI-Q?qs=Wj%2FVkw3K%252BMD4hdUJJxGixg%3D%3D .

Any update? Being curious / n00b: once the pull request from @Urhengulas is merged, does it mean that there will be full support from Rust for all functionalities, or will this be only basic support, with more work needed to get different peripherals / advanced functions to work? :)

Urhengulas, 2022-08-08:
> Any update? Being curious / n00b: once the pull request from @Urhengulas is merged, does it mean that there will be full support from Rust for all functionalities, or will this be only basic support, with more work needed to get different peripherals / advanced functions to work? :)

As far as I understand (which might be wrong) this repository only generates "device support crates" aka. "peripheral access crates" (PAC). After my PR lands we still needs to be build the "hardware abstraction layer" (HAL), for it to be actually useful.

jerabaul29, 2022-08-08:
Aaah, ok, many thanks for the explanation, and I guess the HAL is the more labor intensive part, right? Do you know if there are plans / ongoing work for a dedicated HAL for this chip once your PAC PR lands? :)

Urhengulas, 2022-08-08:
> Aaah, ok, many thanks for the explanation, and I guess the HAL is the more labor intensive part, right? Do you know if there are plans / ongoing work for a dedicated HAL for this chip once your PAC PR lands? :)

I'd be generally interested to participate in that effort. Mainly to try out building support for a new chip. But as you can see from how long my PR is already dangling, my bandwith is currently rather limited 🤷🏾

jerabaul29, 2022-08-08:
Yes, that is very natural, and it is very kind of you to contribute so much already :) . Curious: would the HAL be on this repo too, or on another repo? If a new repo gets set up for the HAL, can you let me know? :)

Btw, @gauteh , guess this is related to what you were mentioning about quite some work needed to take the chip into use? :)

gauteh, 2022-08-08:
The HAL is in different repository, e.g. https://github.com/stm32-rs/stm32l4xx-hal. There is usually significant overlap between the chip families, which can be used when setting up a new HAL. But as far as I can see it is a bit hard to share code between the different families now, so keeping things consistent and fixing common issues requires a bit of effort.

graphex, 2022-11-10:
What's the status of the PR right now @Urhengulas? I'm a n00b but have played with a growing variety of microcontrollers in rust so there is an outside chance I might be able to help get this across the finish line. Maybe after the PAC is in place we can get some help from [this repo's maintainer](https://github.com/David-OConnor/stm32-hal/issues/36) on a working HAL?

Urhengulas, 2022-11-27:
> Maybe after the PAC is in place we can get some help from [this repo's maintainer](https://github.com/David-OConnor/stm32-hal/issues/36) on a working HAL?

Sounds like a plan. Due to time constraints, I couldn't test the PR yet. But I am hopeful I can do that in the following week(s).

Dirbaio, 2023-01-17:
for anyone looking to use Rust on STM32u5: [embassy-stm32](https://github.com/embassy-rs/embassy) already has full support.

TjRichmond, 2024-03-04:
If a crate depends on [embedded-hal](https://crates.io/crates/embedded-hal) to work, does that mean this issue needs to finish supporting the STM32U5 family? 

For example, can various driver crates such as [max7219](https://github.com/almindor/max7219) use embassy-stm32 to solve this, or is our only option updating stm32-rs and beginning stm32u5xx-hal?

burrbull, 2024-03-04:
This project provides only peripheral (register) access level crates (using SVD files as source) and it does not depend on embedded-hal. So you need `stm32u5xx-hal` or some other `hal` to be built on top of this crate.

`embassy` is independent project. It uses its own `pac`s.

Dirbaio, 2024-03-04:
@TjRichmond `embassy-stm32` implements the `embedded-hal` traits, so you can use drivers like `max7219` with it out of the box.

closed 2025-05-27 by burrbull