stm32-rs/stm32-rs #515: Are GigaDevice GD32 Cortex-M MCUs in scope for this repository?

qwandor, 2021-03-16:
Hello,
I'm interested in using Rust on a GD32F130 series microcontroller. As far as I can see there are currently no device or HAL crates to support it. However, the GD32F series seem to peripherals and register layouts mostly compatible with the corresponding STM32 families, apart from some differences in things like clock speed or number of USARTs, so it seems likely that a lot of the existing code from the crates in this repository and the HAL crates could be reused.

Given this, would it make sense to work on supporting them within this same repository? For example, to support the GD32F130 with a feature on the `stm32f1` and `stm32f1xx-hal` crates.

For reference, https://www.gigadevice.com/wp-content/uploads/2021/03/2021-GigaDevice-Selection-Guide_Final.pdf has an overview of the device families, and http://www.gd32mcu.com/en/download/ has datasheets (and SVD files within the addons e.g. GigaDevice.GD32F1x0_DFP.3.1.0.pack).

adamgreig, 2021-03-23:
Closing, see https://github.com/stm32-rs/stm32-rs/pull/520#issuecomment-805302011 for rationale.

closed 2021-03-23 by adamgreig

qwandor, 2021-04-07:
For reference in case anyone else finds this issue, I ended up creating a new repository for this, based on the infrastructure of this one: https://github.com/qwandor/gd32-rs