stm32-rs/stm32-rs #516: STM32L433 RCC AHB1 CRC Enable bit incorrect

cyrusmetcalf, 2021-03-17:
The bit set in RCC->AHB1 for crc enable is shifting (1 << 11).  Needs to be shifting (1 << 12)   according to CMSIS implementation. 

from STM32L4x3.svd
```
            <field>
              <name>CRCEN</name>
              <description>CRC clock enable</description>
              <bitOffset>11</bitOffset>
              <bitWidth>1</bitWidth>
            </field>
```
from CMSIS/Device/ST/STM32L4xx/Include/stm32l433xx.h
```
#define RCC_AHB1ENR_CRCEN_Pos (12U)
#define RCC_AHB1ENR_CRCEN_Msk(0x1UL << RCCAHB1ENR_CRCEN_Pos)
#define RCC_AHB1ENR_CRCEN   RCC_AHB1ENR_CRCEN_Msk
```

cyrusmetcalf, 2021-03-18:
https://github.com/stm32-rs/stm32-rs/pull/517

hellow554, 2021-08-29:
the PR was merged, please close this issue :)

closed 2021-08-29 by adamgreig