stm32-rs/stm32-rs #522: STM32F4 some `bits()` functions are unsafe, others aren't

jonnyando, 2021-03-24:
In the `STM32F446` device at least, I've seen register access via the `bits()` function differ in un-safety between registers and peripherals with no (apparent) rhyme or reason.

For example, the `DATLEN_W` bit field's `bits()` function is `unsafe`, however the `I2SSTD_W` bit field's `bits()` function does not include `unsafe` - both within the same `I2SCFGR` register of the Spi1 peripheral.

I cannot see any distinguishing definition in the SVD causing some to be unsafe and others not, so if this is by design and I'm missing something I apologise.

adamgreig, 2021-03-24:
This is due to svd2rust: it marks `bits()` methods on fields safe if and only if all possible bit values for the field are covered by either a `writeConstraint` range or there is an `enumeratedValue` for every possible pattern, and similarly for entire registers if every bit in the register is covered.

The SVD files from ST don't have either `writeConstraint` or `enumeratedValues`, so out-the-box every `bits()` method is unsafe; they become safe once those extra details have been added by patches in stm32-rs, but only for fields where all possible patterns become covered. For example, `DATLEN` may not be written as `0b11` (the reference manual describes this value as "Not allowed"), so it will never have a safe `bits()` method (since that would allow you to write this not-allowed value). Conversely, `I2SSTD` can be written with all four possible values, and all four are patched in [here](https://github.com/stm32-rs/stm32-rs/blob/master/peripherals/spi/spi_I2S.yaml#L17-L21), so the method becomes safe.

If possible you should use the methods generated for the field variants, like `w.datlen().SixteenBit()` (which is safe) instead of the `bits()` methods.

On an aside, this behaviour is a little controversial - is it really Rust-unsafe to write 11 to that field?  or what if someone added a writeConstraint to a DMA address register field (which might be perfectly acceptable to have in the SVD), but then svd2rust allowed you to write any address to it safely, bypassing memory safety? - but it's what we've got for now; changes would have to be come from [svd2rust](https://github.com/rust-embedded/svd2rust), not here.

I hope that helps clarify the situation!

jonnyando, 2021-03-24:
Exactly the explanation I was looking for, thank you!

closed 2021-03-24 by jonnyando