stm32-rs/stm32-rs #523: is_match_() in stm32l0x1 i2c1 ISR, why the underscore at the end

kuon, 2021-03-25:
I noticed that `is_match()` is called `is_match_()` in i2c isr register.

Is it a typo or is it intended?

https://docs.rs/stm32l0/0.13.0/src/stm32l0/stm32l0x1/i2c1/isr.rs.html#479

wallacejohn, 2021-03-27:
I'm pretty sure this is a result of the [`to_sanitized_snake_case()`](https://github.com/rust-embedded/svd2rust/blob/master/src/util.rs) function in `svd2rust`. Because `match` is a Rust keyword, any occurrences of `match` in an SVD file get an underscore appended to them to avoid potential conflicts.

adamgreig, 2021-03-27:
> I'm pretty sure this is a result of the [`to_sanitized_snake_case()`](https://github.com/rust-embedded/svd2rust/blob/master/src/util.rs) function in `svd2rust`. Because `match` is a Rust keyword, [...]

:+1:, this is it. You might argue it shouldn't do this for `is_match()` since that's an allowed function name, though.

kuon, 2021-03-28:
I opened an issue upstream. <https://github.com/rust-embedded/svd2rust/issues/497>

closed 2021-03-28 by kuon