stm32-rs/stm32-rs #527: Intellisense/autocomplete with rust-analyzer fails to deduce type of register read/modify/write callback

timblakely, 2021-03-27:
Heya!

Loving the project so far (came from C++ and STM's, erm, underwhelming LL libs ;). 

With `rust-analyzer` and VSCode, I'm seeing a failure to deduce the callback parameter types from individual register `.(read|modify|write)(| | ...)` functions:

[image]

Compiles fine though using nightly. Any ideas on why the deduction fails?

timblakely, 2021-03-27:
Apologies for the noise; looks like an update to the rust-analyzer plugin seems to now be deducing the types just fine.

[image]

closed 2021-03-27 by timblakely

adamgreig, 2021-03-27:
Glad to hear it!