stm32-rs/stm32-rs #530: STM32L4x6 XML document invalid

shymega, 2021-04-02:
Hey,

I was just preparing the repository (in lieu of my patch) and when compiling the SVDs, I've found an incomplete XML document - the file is `svd/stm32l4x6.svd`, and appears to have several unmatched elements.

Not quite sure how to resolve it, I'm not too familiar with the tooling, but I'm just reporting it anyway. Let me know if I can help..

Error:
```
❯ make
svd patch devices/stm32l4x6.yaml
Traceback (most recent call last):
  File "/home/dzr/.local/bin/svd", line 8, in <module>
    sys.exit(svdtools_cli())
  File "/home/dzr/.local/venvs/svdtools/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/dzr/.local/venvs/svdtools/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/dzr/.local/venvs/svdtools/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/dzr/.local/venvs/svdtools/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/dzr/.local/venvs/svdtools/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/dzr/.local/venvs/svdtools/lib/python3.8/site-packages/svdtools/cli.py", line 16, in patch
    svdtools.patch.main(yaml_file)
  File "/home/dzr/.local/venvs/svdtools/lib/python3.8/site-packages/svdtools/patch.py", line 1313, in main
    svd = ET.parse(svdpath)
  File "/usr/lib64/python3.8/xml/etree/ElementTree.py", line 1202, in parse
    tree.parse(source, parser)
  File "/usr/lib64/python3.8/xml/etree/ElementTree.py", line 595, in parse
    self._root = parser._parse_whole(source)
xml.etree.ElementTree.ParseError: no element found: line 50528, column 0
make: *** [Makefile:43: svd/stm32l4x6.svd.patched] Error 1
```

adamgreig, 2021-04-04:
Could you check it didn't somehow get corrupted on extraction? Maybe re-run `svd/extract.sh` manually? I've just checked here and it validates against the SVD schema OK:

```
svd$ md5sum stm32l4x6.svd
e6e6718a601b060c054b472b8256b1fd  stm32l4x6.svd
svd$ xmllint --schema cmsis-svd.xsd stm32l4x6.svd --noout
stm32l4x6.svd validates
```

shymega, 2021-04-04:
Interesting, it works fine when I extract manually. I had used the main Makefile goal, and that seemed to be the problem last night. 

Closing, for now, seeing as it now works. Feel free to reopen if you feel this isn't resolved.

closed 2021-04-04 by shymega