stm32-rs/stm32-rs #542: Different names between crate and official ST doc for some bits of the dsihost vmcr register in stm32f469.

Jsbjr, 2021-04-18:
### The names of the bit fields lpvae and lphfpe of the dsihost vmcr register for stm32f469 differ between the documentation and the crate.
| bit number | [documentation](https://www.st.com/resource/en/reference_manual/dm00127514.pdf)  | [crate](https://stm32-rs.github.io/stm32-rs/STM32F469.html#DSIHOST:DSI_VMCR) |
|----------------|----------------------|---------|
| 11 | LPVAE | LVAE | 
| 13 | LPHFPE | LPHFE|
### DSI_VMCR in the crate: 
[image]

###  ST documentation page 601:
[image]
[image]
[image]

burrbull, 2021-04-18:
It is difference between official ST SVD and official ST manual :)
Anyway you can fix this by adding:
```yaml
DSI:
  DSI_VMCR:
    _modify:
      LVAE:
        name: LPVAE
```
in stm32f469.yaml file

Jsbjr, 2021-04-18:
I tried, but got no luck in building even without applying any changes. I get errors with make svd2rust.
[make_svd2rust_errors.txt](https://github.com/stm32-rs/stm32-rs/files/6331290/make_svd2rust.txt)
Apparently none related to the stm32f469. So I tried `cd stm32f4; cargo build --features stm32f469` and got this error:
```
error[E0583]: file not found for module `generic`
  --> src/stm32f469/mod.rs:16:1
   |
16 | pub mod generic;
   | ^^^^^^^^^^^^^^^^
   |
   = help: to create the module `generic`, create file "src/stm32f469/generic.rs"

error: aborting due to previous error

For more information about this error, try `rustc --explain E0583`.
error: could not compile `stm32f4`
```
Also the change in stm32f469.yaml file should probably be 
```
DSIHOST:
  DSI_VMCR:
    _modify:
      LVAE:
        name: LPVAE
      LPHFE:
        name: LPHFPE

```

burrbull, 2021-04-18:
`svd2rust` 0.18 was released only today
use 0.17 or way a couple of days

Jsbjr, 2021-04-18:
Thanks it works much better now. :)
I can make a pull-request if you like.
I don't know if it worth for such a small change.

burrbull, 2021-04-18:
Of course make.

Jsbjr, 2021-04-18:
Ok I did  #543

closed 2025-05-27 by burrbull