stm32-rs/stm32-rs #545: stm32f407g-disc board::stm32::Periphals don't include DWT

mb, 2021-04-22:
Hi, I am using the [stm32f407g-disc board](https://github.com/stm32-rs/stm32f407g-disc) and want to measure some durations for performance-evaluation via [`get_cycle_count()`](https://docs.rs/stm32f407g-disc/0.4.1/stm32f407g_disc/peripheral/struct.DWT.html#method.get_cycle_count) as descripted in [the peripherals documentation](https://docs.rs/stm32f407g-disc/0.4.1/stm32f407g_disc/peripheral/index.html).

I've used `DWT->CYCCNT` from the [CMSIS library in C](https://github.com/ARM-software/CMSIS_5/blob/3fb0787b299980564393ec1cf12a1c547738db1d/CMSIS/Core/Include/core_cm4.h#L907), so the board generally supports CYCCNT. But the [Peripherals of the stm32f407](https://docs.rs/stm32f4/0.13.0/stm32f4/stm32f407/struct.Peripherals.html) don't include DWT. I've opened this issue in this repository, since I think that the peripheral definitions are located here.

Minimal example of what I am trying to do.

* `Cargo.toml`
```toml
[package]
name = "minimal-example"
version = "0.1.0"
edition = "2018"

[dependencies]
stm32f407g-disc = "0.4"
panic-halt = "0.2"
```
* `src/main.rs`
```rust
#![no_main]
#![no_std]

// Halt on panic
use panic_halt as _;

use stm32f407g_disc as board;
use crate::board::{
    entry,
    hal::prelude::*,
    hal::stm32,
    DWT,
};

#[entry]
fn main() -> ! {
    if let Some(p) = stm32::Peripherals::take() {
        // Set up the system clock. We want to run at 168MHz for this one.
        let rcc = p.RCC.constrain();
        let _ = rcc.cfgr.sysclk(168.mhz()).freeze();

        // enable cyclecount https://docs.rs/stm32f407g-disc/0.4.1/stm32f407g_disc/peripheral/index.html
        p.DWT.enable_cycle_counter();
        let _ = DWT::get_cycle_count();
    }

    loop {}
}
```
* `cargo build --target thumbv7em-none-eabihf` (target can be left out, but an additional error will appear) fails with
```bash
error[E0609]: no field `DWT` on type `stm32f407g_disc::Peripherals`
  --> src/main.rs:23:11
   |
23 |         p.DWT.enable_cycle_counter();
   |           ^^^ unknown field
   |
   = note: available fields are: `RNG`, `DCMI`, `FSMC`, `DBGMCU`, `DMA2` ... and 83 others

error: aborting due to previous error

For more information about this error, try `rustc --explain E0609`.
error: could not compile `minimal-example`

To learn more, run the command again with --verbose.
```

If you give me some pointers, I could try to create a PR myself. Thanks.

adamgreig, 2021-04-22:
Because the DWT is part of the CPU peripherals, it's instead included in the [cortex-m](https://crates.io/crates/cortex-m) crate, [documented here](https://docs.rs/cortex-m/0.7.2/cortex_m/peripheral/struct.DWT.html). You can access it using something like

```rust
let cp = cortex_m::Peripherals::take().unwrap();
let dwt = cp.dwt;
dwt.enable_cycle_counter();
let count: u32 = dwt.get_cycle_count();
```

Hope that helps! I'll close this issue now since it's not something we should provide as part of stm32-rs.

closed 2021-04-22 by adamgreig

marceloboeira, 2021-06-14:
@mb How do I reach out to you in private?

mb, 2021-06-14:
I've sent you an email.