stm32-rs/stm32-rs #547: `make` fails with svd2rust version 0.18.0

jonnyando, 2021-04-24:
`svd2rust` recently released version `0.18.0` which contains some breaking changes. This is the version downloaded and installed when running `cargo install svd2rust`.
When running `make` for this crate using this version I am hit with this error:

```
python3 scripts/makecrates.py devices/ -y --families stm32f0 stm32f1 stm32f2 stm32f3 stm32f4 stm32f7 stm32h7 stm32l0 stm32l1 stm32l4 stm32l5 stm32g0 stm32g4 stm32mp1 stm32wl stm32wb
Going to create/update the following directories:
stm32f0/, stm32f1/, stm32f2/, stm32f3/, stm32f4/, stm32f7/, stm32g0/, stm32g4/, stm32h7/, stm32l0/, stm32l1/, stm32l4/, stm32l5/, stm32mp1/, stm32wb/, stm32wl/
mkdir -p stm32f0/src/stm32f0x8
cd stm32f0/src/stm32f0x8; svd2rust -g -i ../../../svd/stm32f0x8.svd.patched
[ERROR svd2rust] In device `STM32F0x8`

    Caused by:
        0: In peripheral `RCC`
        1: In register `CFGR`
        2: Parsing field #8
        3: In field `MCO`
        4: Value 8 out of range 0..8
Makefile:81: recipe for target 'stm32f0/src/stm32f0x8/mod.rs' failed
```

Reinstalling version `0.17.0` of `svd2rust` and rerunning `make` works as expected.

adamgreig, 2021-04-24:
Thanks for reporting! As you mentioned, this isn't a surprise due to a number of changes in svd2rust; stm32-rs is really tied to one version of svd2rust at a time and we haven't updated yet. There's some open issues and PRs (#539, #540, #544, #546) that starts to solve the problems, so hopefully fixed fairly soon. I'll leave this issue open until it's resolved though to help anyone else running into it.

ElektroVirus, 2021-06-23:
I also had problems and found out my svd2rust version is 0.19.0. Its logical that it also fails but I still wanted to document it here.

```

$ make svd2rust
python3 scripts/makecrates.py devices/ -y --families stm32f0 stm32f1 stm32f2 stm32f3 stm32f4 stm32f7 stm32h7 stm32l0 stm32l1 stm32l4 stm32l5 stm32g0 stm32g4 stm32mp1 stm32wl stm32wb
Going to create/update the following directories:
stm32f0/, stm32f1/, stm32f2/, stm32f3/, stm32f4/, stm32f7/, stm32g0/, stm32g4/, stm32l0/, stm32l1/, stm32l4/, stm32l5/, stm32mp1/, stm32wb/, stm32wl/, stm32h7/
mkdir -p stm32f0/src/stm32f0x0
cd stm32f0/src/stm32f0x0; svd2rust -g -i ../../../svd/stm32f0x0.svd.patched
[ERROR svd2rust] In device `STM32F0x0`
    
    Caused by:
        0: In peripheral `RCC`
        1: In register `CFGR`
        2: Parsing field #8
        3: In field `MCO`
        4: Value 8 out of range 0..8
make: *** [Makefile:82: stm32f0/src/stm32f0x0/mod.rs] Error 1
```

```
svd2rust --help
svd2rust 0.19.0 ( )
Generate a Rust API from SVD files
...
```

closed 2021-08-16 by adamgreig