stm32-rs/stm32-rs #548: stm32g0b1 missing RCC registers

timothyhollabaugh, 2021-04-27:
It looks like the new stm32g0b1, b0, and c1 are missing definitions for the RCC peripheral registers. The svd files are missing them as well. These new devices share a reference manual with the older stm32g0 devices, so I would assume it should be there in the same place.

Would it work to copy the RCC section from the stm32g071 svd file? Would a change this big make sense to put in the yaml files?

timothyhollabaugh, 2021-04-27:
Actually, it looks like quite a few peripherals are missing from the SVD files, at least for the stm32g0b1.

I went ahead and extracted the list of peripherals with

`cat stm32g0b1.svd | rg -A 1 '<peripheral>' | tr -d ' \t' |rg '<name>' | sort`

which gives

```
<name>CRC</name>
<name>DAC</name>
<name>DBG</name>
<name>DMAMUX</name>
<name>GPIOA</name>
<name>GPIOB</name>
<name>HDMI_CEC</name>
<name>I2C1</name>
<name>IWDG</name>
<name>LPTIM1</name>
<name>TIM14</name>
<name>TIM15</name>
<name>TIM16</name>
<name>TIM1</name>
<name>TIM2</name>
<name>TIM6</name>
<name>USART1</name>
<name>VREFBUF</name>
<name>WWDG</name>
```

Doing the same for the stm32g071 gives

```
<name>ADC</name>
<name>COMP</name>
<name>CRC</name>
<name>DAC</name>
<name>DBG</name>
<name>DMAMUX</name>
<name>DMA</name>
<name>EXTI</name>
<name>FLASH</name>
<name>GPIOA</name>
<name>GPIOB</name>
<name>HDMI_CEC</name>
<name>I2C1</name>
<name>IWDG</name>
<name>LPTIM1</name>
<name>LPUART</name>
<name>MPU</name>
<name>NVIC</name>
<name>PWR</name>
<name>RCC</name>
<name>RNG</name>
<name>RTC</name>
<name>SCB</name>
<name>SPI1</name>
<name>STK</name>
<name>SYSCFG_VREFBUF</name>
<name>TAMP</name>
<name>TIM14</name>
<name>TIM15</name>
<name>TIM16</name>
<name>TIM1</name>
<name>TIM2</name>
<name>TIM6</name>
<name>UCPD1</name>
<name>USART1</name>
<name>WWDG</name>
```

Diffing the two shows that the stm32g0b1 is missing the ADC, COMP, DMA, EXTI, FLASH, LPUART, MPU, NVIC< PWR, RCC, RNG, RTC, SCB, SPI1, STK, TAMP, and UCPD1 peripherals. Th b1 seems to be almost a superset of the 71, so I would expect all or most of the 71's peripherals to be in the b1. Especially important ones like DMA and FLASH. The stm32g0b1 also includes a USB peripheral, so I would expect that as well.

I'm not sure what to make of all the missing peripherals from the SVD files. Perhaps they only included the peripherals that changed from the 71?

adamgreig, 2021-04-27:
Oh, that's a pain. I wonder why they bothered with so many SVD files if they don't even include all the peripherals!

One option is to copy the peripherals from the other SVD files, which the patcher can do for you:

```yaml
_copy:
    TIM1:
        from: ../svd/stm32f302.svd:TIM1
```

This also copies all the interrupts from that peripheral in the other SVD, so it would be worth double checking they don't end up messed up, and it copies from the original SVD, so any patches the other device applies would need to be applied here too.

Another option would be to just use the same source SVD file and patch it differently for the different devices (or even merge the devices...) but that's probably more work and harder to figure out, so maybe not the best choice.

timothyhollabaugh, 2021-05-05:
So I gave it a show by manually copying the RCC section over from the stm32g071 SVD, and that works good. Next I'll try doing that with the patcher.

greglapt, 2021-07-15:
ST made a new svd file for stm32g0b1.

https://community.st.com/s/question/0D53W00000uYrEKSA0/stm32cubeide161-sfrs-doesnt-show-usb-or-fdcan-registers
https://community.st.com/sfc/servlet.shepherd/document/download/0693W00000BcPyQQAV?operationContext=S1

Dirbaio, 2022-01-24:
#666 fixes g0b1, g0c1, but still missing in g0b0

greglapt, 2022-04-28:
You can take svd files for g050, g051, g061 from STM32CubeIDE 1.9.0
STM32G050.svd 1164821 bytes
STM32G051.svd 1178993 bytes
STM32G061.svd 1178993 bytes
Path of it (on Windows): c:\ST\STM32CubeIDE_1.9.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.productdb.debug_2.0.200.202201061234\resources\cmsis\STMicroelectronics_CMSIS_SVD\