stm32-rs/stm32-rs #561: ADC/linker problem - won't compile (STM32L432KC)

nebelgrau77, 2021-06-01:
I'm trying to use the ADC example with an STM32L432KC, using the latest version from GH ("git = "https://github.com/stm32-rs/stm32l4xx-hal"). It won't compile, throwing the following error:

```
note: "rust-lld" "-flavor" "gnu" "--eh-frame-hdr" "-L" (omitted)
= note: rust-lld: error: undefined symbol: ADC1_2
>>> referenced by stm32l4.arnfbrxu-cgu.3
>>> stm32l4-3c82b24cdd335e86.stm32l4.arnfbrxu-cgu.3.rcgu.o:(__INTERRUPTS) in archive
```

As suggested in the L4 section, I cross-checked between PAC and Reference Manual:

in [PAC ](https://github.com/stm32-rs/stm32-rs/blob/master/devices/stm32l4x2.yaml) there are the following lines:

```json
ADC1:  
    _modify:    
        _interrupts:      
            ADC1:        
                name: ADC1_2
```
and ADC1_2 also appears in the NVIC vector table in the [RM](https://www.st.com/resource/en/reference_manual/dm00151940-stm32l41xxx42xxx43xxx44xxx45xxx46xxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) as ADC1 and ADC2 global interrupt (page 322). 

So it does seem to be all in order, yet it won't work. What am I doing wrong?

korken89, 2021-06-01:
I had a look now an am unable to see where this goes wrong.
Could you share an example that exercises the error?

nebelgrau77, 2021-06-01:
I basically tried to replicate [this example](https://github.com/stm32-rs/stm32l4xx-hal/blob/master/examples/adc.rs), but I will upload my actual code later.

korken89, 2021-06-01:
Thanks, please post here when you have it available

nebelgrau77, 2021-06-02:
https://github.com/nebelgrau77/STM32L4_ADC_test - I just put together a simple ADC to UART code, following the example from the STM32L4 crate.