stm32-rs/stm32-rs #605: RCC H735 conflicting rules

burrbull, 2021-08-28:
Rules in stm32h735 (RM0468) are partially incompatible with that in `common_patches/h7_common_singlecore.yaml`.
```console
svdtools makedeps devices/stm32h735.yaml .deps/stm32h735.d
In String("name"): conflicting rules String("OCTOSPISEL") and String("QSPISEL"), ignored
In String("name"): conflicting rules String("SWPMISEL") and String("SWPSEL"), ignored
In String("name"): conflicting rules String("I2C1235SEL") and String("I2C123SEL"), ignored
In String("name"): conflicting rules String("USART16910SEL") and String("USART16SEL"), ignored
```

richardeoin, 2021-08-30:
It seems that the correct rules (the ones in `devices/stm32h735.yaml`) are used, but indeed the current situation is ambiguous and should be fixed. I'm a bit confused on a couple of things:

* I think the conflicting rules are in `common_patches/h7_rcc_src_sel.yaml`. Not sure on the connection to `common_patches/h7_common_singlecore.yaml`
* How do I reproduce the log above? Is the released svdtools 0.1.16 enough, or do I need master or some other branch?

burrbull, 2021-08-31:
I've found this using rust version of `svd patch`: https://github.com/burrbull/svdtools/tree/riir2/src

The problem is `yaml_includes` function which is not smart enough to find duplicates or conflicts in rules.

commit f8ba54ff (2021-09-03): H735: fix conflicting rules and one duplicate rule

closes #605
commit 13d1e705 (2021-09-03): Merge #615

615: H735: fix conflicting rules and one duplicate rule r=adamgreig a=richardeoin

closes #605
PR #615 (2021-09-03): H735: fix conflicting rules and one duplicate rule
closes #605
closed 2021-09-03 by bors[bot]