stm32-rs/stm32-rs #618: H7 xCR registers have errors

David-OConnor, 2021-09-04:
It looks like the variant used for WB is correct, but it's also teh same on H7. Specifically: The H7 PAC (probably SVD) has xCR bit 19 as NODIV, and 27 as MCKEN. RM states 27 is not used, and 19 is NOMCK.

Of note, the SAI periph seems similar or identical other than this, and NODIV and MCKEN both relate to generating a master clock.

adamgreig, 2021-10-02:
It looks like all our STM32H7 SVDs have the exact same SAI registers and fields, and they're all pretty badly wrong compared to the RMs. All the B registers are missing, and the A registers are all offset by 4, so CR1 aliases GCR.

In terms of this specific problem though: many of the H7 RMs do have 19 as NODIV and 27 as MCKEN, I think only the 743/753 (RM0433) has NOMCK nistead. As far as I can tell this is a genuine difference, not an RM error, and the other H7 and WB probably use a newer version of the SAI peripheral.

adamgreig, 2021-10-03:
This was renamed in #347 to deliberately go against the RM in preference of consistency with the other devices. That said, looking at the RMs I'm not clear if NODIV and NOMCK are the same:

[image]

[image]

[image]

@richardeoin, any thoughts?

David-OConnor, 2021-10-03:
Good find, and my reaction was that we cannot assume it's the same.

richardeoin, 2021-10-03:
What's happening here is that [AB]CR bit 19 has two related functions, and each RM picks a different one from those two functions to use in the bit description. In fact, the bit has both functions on all H7 parts.

## RM0433

||Function 1 | Function 2
|---|---|---
| bit 19 NOMCK = 0 | master clock enable = 1 | frequency ratio between FS and MCLK is fixed to 256 or 512 (RM0433 rev 7 p2266 bottom paragraph) |
| bit 19 NOMCK = 1 | master clock enable = 0 | frequency ratio depends on FRL |

## RM0399 RM0455 etc

||Function 1 | Function 2
|---|---|---
| bit 19 NODIV = 0 | master clock enable = 1 | frequency ratio between FS and MCLK is fixed to 256 or 512 (RM0399 rev 3 p2422 first bullet point) |
| bit 19 NODIV = 1 | master clock enable = `MCKEN` | frequency ratio depends on FRL (RM0399 rev 3 p2422 second bullet point) |

The "Audio block clock generator overview" figures in each RM show how NOMCK/NODIV is used twice.

richardeoin, 2021-10-03:
Since this is terribly confusing, I'm thinking the best way forward is to match the RM bit naming exactly (so different names for bit 19) but write a better bit description that applies both to NOMCK, NODIV

David-OConnor, 2021-10-03:
How does bit 27 play into this? It's missing from the H7 RM entirely.

adamgreig, 2021-10-03:
There's four H7 reference manuals: RM0399, RM0433, RM0455, and RM0468, covering different H7s: bit 27 MCKEN is only missing from RM0433, but it's present in the others. When it's missing (in RM0433 devices i.e. 742, 743/53, 750), bit 19 (NOMCK) disables MCK entirely if set, and enables it if cleared, whereas with the other H7s, bit 19 (NODIV) delegates control to bit 27 MCKEN if set, which can then enable or disable it, but if NODIV is clear, MCK is always enabled.

I'm just summarising @richardeoin's comment above: basically, bit 19 does the same thing on all H7s, even though the name changes, because some newer devices also add bit 27 MCKEN, which works together with bit 19 to potentially disable MCK.

commit 4b681370 (2021-10-03): H7 RM0433: Rename SAI [AB] CR bit 19 to NOMCK and remove MCKEN

Other H7 parts are unchanged. Improve description of NOMCK/NODIV bit

See previous PR #347. Fixes #618
commit a75f1df3 (2021-10-03): Merge #640

640: H7 RM0433: Add NOMCK at SAI [AB]CR1 bit 19 and remove MCKEN r=adamgreig a=richardeoin

Other H7 parts are unchanged. Improve description of NOMCK/NODIV bit

See previous PR #347. Fixes #618
PR #640 (2021-10-03): H7 RM0433: Add NOMCK at SAI [AB]CR1 bit 19 and remove MCKEN
Other H7 parts are unchanged. Improve description of NOMCK/NODIV bit

See previous PR #347. Fixes #618
closed 2021-10-03 by bors[bot]