stm32-rs/stm32-rs #619: Tracking issue for 0.14

adamgreig, 2021-09-06:
I plan to release 0.14 soon, once a few pending PRs are sorted. Currently the list is:

* [ ] #606
* [ ] #612 
* [ ] #613
* [ ] Possibly #593 (@maximeborges, do you think it's worth trying to get it in soon, or OK to wait?)

I'll try and go through the other PRs which I think are mostly waiting on author, and for simple fixes I'll try and get them in, but they won't block getting 0.14 out the door, it's waited long enough. I'll put together the changelog updates this week.

If there are any open issues you'd like fixed, please comment about them here and/or submit a PR!

maximeborges, 2021-09-06:
For me there is still a lot of work to do for [Add proper support for STM32MP1 #593](https://github.com/stm32-rs/stm32-rs/pull/593) so I'd say that it can wait. If other people want to help, we can work together on a branch for this one, but I'm not sure it would be useful to anyone in the current state.
I might work a bit more on RP2040 support in the next few weeks after those PRs, depending on what I can do with it for a project, otherwise I will come back to the F7 family.

maximeborges, 2021-09-07:
Adding [[stm32f7] Fix SDMMC registers](https://github.com/stm32-rs/stm32-rs/pull/620)

PR #638 (2021-10-03): Prepare for v0.14.0 release
Closes #619. I plan to get #637 in, have a brief check over recent open issues, then release this today or tomorrow. Sorry for the wait!
closed 2021-10-03 by adamgreig