stm32-rs/stm32-rs #629: 16 bit fields needing 12-bit integers to write?

David-OConnor, 2021-09-17:
Hi. When I try to write a DAC register that's only 12 bits (eg `DAC_DHR12R1`), the PAC requires a u32. It seems like a u16 would make more sense. Thoughts?

burrbull, 2021-09-18:
My thought that this is strange question. You even did not write what chip model you are saying about. Please link where it requires u32.

adamgreig, 2021-09-18:
This is probably more of an svd2rust question. Are you trying to write to the whole register, or to the field inside it? If the field is only 12 bits then svd2rust should generate a function that takes a 16-bit value, but since the whole register is 32 bits wide, writing to the whole register (which would presumably require unsafe in this case) takes a 32-bit value.

David-OConnor, 2021-09-18:
Thank you. I need to do more research on this. 

When I do a write like this:
```rust
regs.dac_dhr8r2.modify(|_, w| unsafe { w.bits(val as u32) })  // regs is the PAC reg block for DAC.
```
I have to convert from u16 to u32, or I get an error to that effect. When I look up the docs for the relevant fields, it shows `bits` as taking a u16. Eg: https://docs.rs/stm32h7/0.13.0/stm32h7/stm32h743v/dac/dhr12r2/struct.DACC2DHR_W.html

adamgreig, 2021-09-18:
In that case you're writing to the whole register, rather than a field within it. You could try:

```rust
regs.dac_dhr8r2.modify(|_, w| w.dacc2dhr().bits(val));
```

Might still need unsafe if that field isn't set to allow any value to be written, not sure. The difference is in selecting the specific field in the register to write, even though in this case there's only one field, because the register itself is 32 bits wide, only the field is 12 bits.

David-OConnor, 2021-09-18:
Oh, nailed it! Thanks!

closed 2021-09-18 by David-OConnor