stm32-rs/stm32-rs #636: H743 and H743 DFSDM offsets wrong

David-OConnor, 2021-10-02:
The offsets appear to be correct for all other H743 variants. They all appear to be wrong on H743 and H743V. Working on a fix, but can't get changes to take ATM.

adamgreig, 2021-10-02:
Could you push a branch that shows them not working? It's hard to tell from just that snippet, which looks OK in isolation.

David-OConnor, 2021-10-02:
Done - https://github.com/stm32-rs/stm32-rs/pull/637

David-OConnor, 2021-10-02:
I think the "not working" is due to typos in the reg names I added.

adamgreig, 2021-10-02:
If nothing else, we usually strip the peripheral name from field names using the `_strip` command for things like this ([example](https://github.com/stm32-rs/stm32-rs/blob/master/devices/common_patches/sdio_f446.yaml#L12)).

David-OConnor, 2021-10-02:
Added strip, and the address offsets appear to have taken. Leaving as draft until further QC and ops testing. Appreciate it!

adamgreig, 2021-10-02:
Nice! Annoyingly it seems like the names such as `DFSDM3_CR2` get turned into plain `3_CR2` which is not valid (cannot start with a number). Looks like that's only an issue on 743/743v, so perhaps that patch file could also rename those registers to the `FLT3CR2` name used elsewhere.

David-OConnor, 2021-10-02:
That would be idea; make it all match RM.

Of note, it seems like it's prefixing `_` to make it valid, but I'm running into something very weird where it's expecting 2 of the regs (`_1_FCR` and `_3_FCR` to be methods (?) based on compile errors, and being ommitted from the top-of-the-module list of offsets. Not sure what's up. Could just make all the names match the RM.

adamgreig, 2021-10-02:
`svd2rust` will prefix `_`, but the generated SVD file fails validation against the SVD specification because of the invalid names. Since you're already having to put the new address offset for the affected registers, you could also put `name:` in that same block to rename them.

If it's turning registers into methods, that's typically because they overlap or have the same address (like CCMR input/output mode registers on STM32 timers), so they are taken out of the register block and the methods are used instead.

David-OConnor, 2021-10-02:
Names fixed across H7

adamgreig, 2021-10-02:
Fixed by #637

closed 2021-10-02 by adamgreig