stm32-rs/stm32-rs #641: G0 DMA IFCR 0.14 reversion

David-OConnor, 2021-10-04:
[0.14](https://docs.rs/stm32g0/0.14.0/stm32g0/stm32g081/dma/ifcr/index.html)
[0.13](https://docs.rs/stm32g0/0.13.0/stm32g0/stm32g081/dma/ifcr/index.html)
This was correctly AFAIK in 0.13, but now the fields aren't right.

burrbull, 2021-10-04:
Are problems in patching or svd2rust?

burrbull, 2021-10-04:
Looks like original ST SVDs are incorrect.

adamgreig, 2021-10-04:
Thanks for the bug report!

This definitely looks like a regression, I suspect from #514 but then not caught by the memory map comparison because it can't work on PRs with updated SVD files. It seems ST's SVDs for many G0s originally had this error, and it was patched in #398, but reverted in #514 because the new SVD files fixed the error. However, it looks like not all the SVD files had the error fixed (G070 and G081) and hence the wrong field values have returned.

I'll try and get this fixed and do a patch release of stm32g0 soon, though if anyone else wants to have a go at a fix please go ahead!

DerFetzer, 2021-11-20:
The current SVD release on the ST website (https://www.st.com/resource/en/svd/stm32g0_svd.zip) seems to include the missing peripherals.
@adamgreig Are you working on this already or should I give it a go? This would be my first contribution to this repo. Are there any common pitfalls I should be aware of?

nickmertin, 2022-12-05:
Did #666 fix this?

dimpolo, 2022-12-12:
> #514 seems to have some missing registers for the new variants. [docs](https://docs.rs/stm32g0/0.14.0/stm32g0/stm32g0b0/index.html) Registers like `RCC` , `PWR`, `GPIOA` or `FLASH` are not available.

This is still true ([current docs](https://docs.rs/stm32g0/0.15.1/stm32g0/stm32g0b0/index.html)) and ST's SVD haven't been updated since :/
I opened an issue with them a while ago, but never got a response.
Does someone have a good connection to them?

burrbull, 2025-02-09:
Should be fixed. Reopen if it is not.

closed 2025-02-09 by burrbull