stm32-rs/stm32-rs #642: ``cortex-m-rt`` dependency error

NathanMcMillan54, 2021-10-04:
Pretty much every version of cortex-m-rt before [v0.6.15](https://crates.io/crates/cortex-m-rt/0.6.15) has been yanked, and I know lots of libraries in this organization depend on it. I'm just letting anyone who has had or will have a problem compiling any of these libs know that that is probably why.

adamgreig, 2021-10-04:
Have you had an issue? The reason is detailed in https://github.com/rust-embedded/cortex-m-rt/issues/334 but since there's still a version up for 0.5, 0.6, and 0.7, I wasn't expecting any problems to come up (and it did solve a number of confusing linker issues).

NathanMcMillan54, 2021-10-04:
Sorry! I should've showed the error:
```command line
error: failed to select a version for `cortex-m-rt`.
    ... required by package `stm32f4xx-hal v0.8.3`
    ... which is depended on by `stm32-eth v0.2.0`
    ... which is depended on by `a_crate v0.1.0 (path_to_a_crate)`
    ... which is depended on by `another_crate v0.1.0 (path_to_another_crate)`
    ... which is depended on by `another_other_crate v0.1.0 (path_to_another_other_crate)`
versions that meet the requirements `^0.6.10` are: 0.6.15

the package `cortex-m-rt` links to the native library `cortex-m-rt`, but it conflicts with a previous package which links to `cortex-m-rt` as well:
package `cortex-m-rt v0.7.0`
    ... which is depended on by `a_crate v0.1.0 (path_to_a_crate)`
Only one package in the dependency graph may specify the same links value. This helps ensure that only one copy of a native library is linked in the final binary. Try to adjust your dependencies so that only one package uses the links ='cortex-m-rt' value. For more information, see https://doc.rust-lang.org/cargo/reference/resolver.html#links.

failed to select a version for `cortex-m-rt` which could resolve this conflict

```

NathanMcMillan54, 2021-10-04:
This has nothing to do with linker errors

adamgreig, 2021-10-04:
It sounds like this doesn't have anything to do with the recent yanks: without the yanks, Cargo would have selected 0.6.14 _and_ 0.7.0, which would _then_ give you a linker error, whereas with the yanks it gives you a more helpful error: it cannot satisfy your cortex-m-rt requirements, because `a_crate` requires 0.7.0, and `stm32f4xx-hal` requires 0.6.X, and you cannot have both in one binary.

The solution is to use 0.6.X (or `>=0.6.15,<0.8`) in `a_crate`, and once `stm32f4xx-hal` releases its next version, it should be compatible with `>=0.6.15,<0.8` itself, permitting use of 0.7.

NathanMcMillan54, 2021-10-04:
Oh ok thanks!

closed 2021-10-04 by NathanMcMillan54