stm32-rs/stm32-rs #670: OTG_HS interrupt possibly missing from stm32h7xx_hal::interrupt enum for STM32H723ZG

TomDeRybel, 2021-12-03:
I'm porting over some code to a ST Nucleo-H723ZG board and ran into a little snag, shown below. The other interrupts used in the code compile fine (eg: EXTI15_10, TIM1_UP).

The OTG_HS interrupt and friends are listed on the documentation website and worked fine on the H743 I used before.
https://docs.rs/stm32h7xx-hal/latest/stm32h7xx_hal/enum.interrupt.html#variant.OTG_HS

However, none of the USB interrupts show up in the documentation for the STM32H735 crate, generated with cargo doc.

`error[E0599]: no variant or associated item named `OTG_HS` found for enum `stm32h7xx_hal::interrupt` in the current scope
   --> src/main.rs:251:17
    |
251 |     #[task(binds = OTG_HS, shared = [usb_dev, usb_cdc])]
    |                    ^^^^^^ variant or associated item not found in `stm32h7xx_hal::interrupt`
`

I'm using the following versions/cargo lines:
`stm32h7 = { git = "https://github.com/stm32-rs/stm32-rs-nightlies", features = ["stm32h735","rt"] }`
`stm32h7xx-hal = { git = "https://github.com/stm32-rs/stm32h7xx-hal", features = ["stm32h735","rt","usb_hs"] }`

As an aside, some register field names seem to have changed between the H743 and H725. Not sure if this is intentional:
H743
`ctx.device.DBGMCU.cr.modify(|_, w| {
			w.dbgsleep_d1().set_bit();
			w.dbgstby_d1().set_bit();
			w.dbgstop_d1().set_bit()
		});`

H723
`ctx.device.DBGMCU.cr.modify(|_, w| {
			w.dbgslpd1().set_bit();
			w.dbgstbd1().set_bit();
			w.dbgstpd1().set_bit()
		});`

Thank you for all the work done on these chips! These crates make them so much more friendly to use.

adamgreig, 2021-12-03:
Thanks for reporting, this looks like an error in the H735 SVD file from ST. If you'd like to give fixing it a go, the fix is to add new interrupts to the `devices/stm32h735.yaml` file in this repository, something like:

```yaml
OTG_HS:
    _add:
        _interrupts:
            OTG_HS:
                value: 77
                description: OTG_HS global interrupt
```

and so forth.

For the field names that changed, that's probably again coming from ST's different SVD files. Ideally they should match whatever's in the reference manual for that chip, so if the H735 SVD doesn't agree with the H735 reference manual, we could update them to match the H743.

TomDeRybel, 2021-12-04:
Thank you for the information, I'll give it a go next week.

TomDeRybel, 2021-12-06:
I've gone through the entire interrupt table in RM0468 (rev.2) on page 735 and compared it to the contents of the relevant mod.rs for the STM32H735 family. There were a rather a lot of missing entries and a few naming issues. 

I've tried to add them all, but can't get most of them accepted by svdtools when running "make patch".  Eg: `svdtools.patch.MissingPeripheralError: Could not find OTG_HS`

My attempts are attached, the changes are confined to the section between dashed lines, at the end of the file, just before the includes start.

[stm32h735.zip](https://github.com/stm32-rs/stm32-rs/files/7661099/stm32h735.zip)

TomDeRybel, 2021-12-06:
Regaring the DBGMCU naming, these are indeed different in the RM0468, page 3199, and match those for the H743.
Should I add those to stm32h735.yaml, or should the rename go elsewhere?

David-OConnor, 2022-06-26:
I may take a crack at this

TomDeRybel, 2022-06-27:
Thank you! Looking forward to replace my amateur fixes by the real deal :-)

richardeoin, 2022-08-15:
This was closed by #677. Thanks @TomDeRybel !

closed 2022-08-15 by richardeoin