stm32-rs/stm32-rs #780: STM32G030 (and maybe other) DMA registers are incorrect

jamesmunns, 2022-10-05:
The current STM32G030 Channel registers are generated into something that looks like this:

```
// dma/ch.rs

#[doc = r"Register block"]
#[repr(C)]
pub struct CH {
    #[doc = "0x00 - DMA channel x configuration register"]
    pub cr: CR,
    #[doc = "0x04 - DMA channel x number of data register"]
    pub ndtr: NDTR,
    #[doc = "0x08 - DMA channel x peripheral address register"]
    pub par: PAR,
    #[doc = "0x0c - DMA channel x memory address register"]
    pub mar: MAR,
}
```

However there are actually *5* words there, there should be a reserved word at the end of each channel field:

[image]

The DMA register hints at this, but the comments don't match the generated code:

```rust
#[doc = r"Register block"]
#[repr(C)]
pub struct RegisterBlock {
    #[doc = "0x00 - low interrupt status register"]
    pub isr: ISR,
    #[doc = "0x04 - DMA interrupt flag clear register"]
    pub ifcr: IFCR,
    #[doc = "0x08..0x6c - Channel cluster: CCR?, CNDTR?, CPAR?, and CMAR? registers"]
    pub ch: [CH; 5],
}
impl RegisterBlock {
    #[doc = "0x08..0x1c - Channel cluster: CCR?, CNDTR?, CPAR?, and CMAR? registers"]
    #[inline(always)]
    pub fn ch1(&self) -> &CH {
        &self.ch[0]
    }
    #[doc = "0x1c..0x30 - Channel cluster: CCR?, CNDTR?, CPAR?, and CMAR? registers"]
    #[inline(always)]
    pub fn ch2(&self) -> &CH {
        &self.ch[1]
    }
    #[doc = "0x30..0x44 - Channel cluster: CCR?, CNDTR?, CPAR?, and CMAR? registers"]
    #[inline(always)]
    pub fn ch3(&self) -> &CH {
        &self.ch[2]
    }
    #[doc = "0x44..0x58 - Channel cluster: CCR?, CNDTR?, CPAR?, and CMAR? registers"]
    #[inline(always)]
    pub fn ch4(&self) -> &CH {
        &self.ch[3]
    }
    #[doc = "0x58..0x6c - Channel cluster: CCR?, CNDTR?, CPAR?, and CMAR? registers"]
    #[inline(always)]
    pub fn ch5(&self) -> &CH {
        &self.ch[4]
    }
}
```

(Each channel is shown to be 20 bytes in the hex, but the first snippet is only 16 bytes per channel).

I'm guessing this affects more than the STM32G030 specifically, but I can verify patching in the following definition manually makes DMA channels other than CH1 work for me:

```rust
#[doc = r"Register block"]
#[repr(C)]
pub struct CH {
    #[doc = "0x00 - DMA channel x configuration register"]
    pub cr: CR,
    #[doc = "0x04 - DMA channel x number of data register"]
    pub ndtr: NDTR,
    #[doc = "0x08 - DMA channel x peripheral address register"]
    pub par: PAR,
    #[doc = "0x0c - DMA channel x memory address register"]
    pub mar: MAR,
    pub _RESERVED: u32,
}
```

I have no idea how the stm32 PAC machinery works, so hopefully this report is useful for y'all :D

burrbull, 2022-10-05:
It is not an issue of generation.
There are incorrect addresses of registers in source SVD.
Need to `_modify` those addresses before collect to cluster.

burrbull, 2022-10-05:
You are right. Need to check svd2rust. Especially `max_cluster_size`

burrbull, 2022-10-05:
This should fix an issue
https://github.com/rust-embedded/svd2rust/pull/661

closed 2022-10-17 by bors[bot]