stm32-rs/stm32-rs #786: Mixing versions of PAC crates - DEVICE_PERIPHERALS multiply defined

mcbridejc, 2022-11-19:
Hi,

I'm trying to create a crate which depends on a PAC crate, and have hit an issue where I can't find a way to do this reliably using the `stm32f0` PAC. 

```
warning: Linking globals named 'DEVICE_PERIPHERALS': symbol multiply defined!

error: failed to load bitcode of module "stm32f0-9c614b73ce91b0b9.stm32f0.38c6535e-cgu.0.rcgu.o": 
```

This error occurs, as I understand it, because both versions are declaring the DEVICE_PERIPHERALS global with `#[no_mangle]`, which makes sense (I'm not sure why the `no_mangle` is needed, but I presume there's a reason). It works fine if app->hal->pac and app->my_crate->pac depend on the same version, but in general I don't see a way to ensure this is the case (even with `stm32f0="*"`, cargo is pulling in the latest available, not combining with the HAL version, and anyway there can and are breaking differences between PAC versions).

Is there any way to achieve what I'm trying to do? If not, does a feature gate for DEVICE_PERIPHERALS make sense in the future? 

Thanks,
-Jeff

adamgreig, 2022-11-20:
In some sense this is more an issue with [svd2rust](https://github.com/rust-embedded/svd2rust) which is what generates the PACs including the `DEVICE_PERIPHERALS` global (it's `no_mangle` as otherwise the name will be mangled to include the PAC version, so two different-versioned PACs could incorrectly co-exist; making it no-mangle causes a linker error instead).

One option is to depend on the HAL instead, and use its `pac` export to access the PAC types; this guarantees you'll be exactly compatible with the HAL even as it updates, though of course if your users want to update HAL to a newer version than you support there'll be problems.

Another option which is fairly commonly done is to bundle your own mini PAC for just the peripheral you're interested in - for example literally copying the relevant files out of the PAC, or generating them yourself with svd2rust (or another generator such as stm32ral which is what https://github.com/stm32-rs/synopsys-usb-otg/tree/master/src/ral does). You'll probably need an `unsafe` method that the end-user calls to construct your types that access the registers directly, where the user promises to not simultaneously access them via the normal PAC method; or you could have a safe method that consumes a type from the HAL, but that has similar issues around versioning.

A feature gate could work - it would be a bit like what I describe above but without literal vendoring - but I don't think it would actually resolve the versioning issues, because the type you'd want your user to pass in from their (newer/older) PAC wouldn't match the type you have internally, so in practice they'd still need to be using the same version.

mcbridejc, 2022-11-20:
Thanks for the response @adamgreig! And, thanks for the pointer to stm32ral. That looks like it is probably what I want, since it has the `nosync` feature -- I wish svd2rust had such a feature! 

I would prefer to accept a peripheral object in my `init()` function but I can live without it, as "don't access TSC registers in your application" is a simple and easy to comply with requirement; well worth the benefit of being able to package a driver for re-usability in multiple projects.

I see your point, that this is an svd2rust issue, and I've just found the RFC where this was laid out as an intentional decision a while back: https://github.com/rust-embedded/svd2rust/issues/157. This makes me sad, a bit.  After some small complaining below (please consider reading optional) I will close this issue, as I see this is not the place. Perhaps I'll open an issue there, although I fear it may not be well received.

Having to copy/paste PAC code seems like a bad pattern, especially since I want to support many different parts (the TSC peripheral is identical among many STM parts). And it ends up with the same requirement on the app developer to ensure they are not also accessing the peripheral used by the library. Depending on a second set of register definitions from stm32ral is better, for sure, but it would be nice to be able to use the same PAC interface (and this solution only applies to STM32).

To me, the PAC -- as basically an equivalent of CMSIS headers -- is too low a level to try to enforce an opinion like this. Rust offers a lot of benefits, and I really want to use it for all of my embedded work, but it sometimes seems that decisions about how to protect inherently unsafe hardware peripherals result in severe usability costs for very small practical gains, and that this makes it harder to take advantage of the best benefits of Rust which is protecting the normal memory access of my programs (as well as a lot of other features). 

For my $0.02, the no_mangle is just unnecessary. But at a minimum, I really think it should be feature-gated in PACs for flexibility.

closed 2022-11-20 by mcbridejc

adamgreig, 2022-11-20:
You're not the only one to think the PAC should maybe be entirely unsafe and the provision of singletons left to HALs; for example that's how things work in the [embassy](https://embassy.dev) project, which uses a forked version of svd2rust called chiptool that doesn't generate singletons. I think svd2rust could probably move in that direction eventually, but no one is working on it at the moment.

We do need the no_mangle `DEVICE_PERIPHERALS` as long as the PAC provides safe access to peripherals, because without it multiple versions of the PAC could be used to obtain multiple overlapping references to the same memory, which can lead to data races which are UB. If the PAC only provided unsafe access, with a safety requirement that you ensure no other simultaneous access to this peripheral happens, then multiple versions could happily coexist, which may well be a price worth paying.

mcbridejc, 2022-11-20:
> We do need the no_mangle DEVICE_PERIPHERALS as long as the PAC provides safe access to peripherals, because without it multiple versions of the PAC could be used to obtain multiple overlapping references to the same memory, which can lead to data races which are UB.

I'm not sure I understand the UB thing. Since the peripheral registers are not referencing *memory*, but registers which may  have all sorts of behaviors, it seems any register access is UB whether there are multiple references or not.  Practically speaking, I do see how any non-atomic read-modify-write is a data race, and those cases can be prevented by ensuring single reference. (I guess I'm just concurring here with the view that all PAC access should be unsafe!)

But on the no_mangle bit... the case that the no_mangle is preventing is so narrow! I'm having a hard time coming up with a plausible scenario where calling `take()` on two different versions happens accidentally. Preventing another PAC version used on a lib crate doesn't ensure no other register ownership if I can safely use the stm32ral crate in my library to access those registers (because it hides the unsafe, just like the PAC). Or I can simply (unsafely) de-reference a pointer to the register.

adamgreig, 2022-11-20:
Because the registers are memory-mapped, accesses to them are treated the same as memory accesses by Rust and LLVM, so we have to avoid UB the same way. If it were the case that any register access is UB we'd be in deep trouble! You're right though; the fact that some other PAC could exist without using a `DEVICE_PERIPHERALS` global does negate the safety promise the current svd2rust PACs make.

I think in practice the likely scenario avoided by the current global is a HAL using one version of a PAC, and an application also pulling in another version of a PAC, and getting to access registers both through the PAC and through the HAL without the HAL knowing about it. Whether this is all that likely or not though, I don't know... so from a practical point of view you can indeed see the appeal to the PACs being unsafe-only and the HALs having an unsafe constructor where you pinky-promise to not access those registers any other way.

> Or I can simply (unsafely) de-reference a pointer to the register.

So long as you have to use `unsafe` to do it, that's OK. The issue is only if you can create UB without using `unsafe`, in which case the library/abstraction is in some sense unsound. Which is the case for using e.g. stm32ral alongside an svd2rust-generated PAC because they don't share the exclusivity mechanism...