stm32-rs/stm32-rs #788: tim.cnt sometimes function, sometimes property

stillinbeta, 2022-12-04:
I'm working on adding stm32g0b1 support to g0xx_hal, and I've noticed something odd:

for e.g. TIM14, the `cnt` attribute [is a field](file:///home/ellie/Projects/keyboards/stm32-rs-nightlies/stm32g0/target/doc/stm32g0/stm32g0b1/tim14/struct.RegisterBlock.html#structfield.cnt)
but for e.g. TIM2, `cnt` is a [method](file:///home/ellie/Projects/keyboards/stm32-rs-nightlies/stm32g0/target/doc/stm32g0/stm32g0b1/struct.TIM2.html#method.cnt). 

Can this be corrected in the YAML or SVDs, or should I try to work around it in the HAL code?

newAM, 2022-12-04:
Inconsistencies are generally corrected in the YAML to make the HAL work easier, the SVDs are just the vanilla release from STM.

The YAML has its own way of patching the SVDs, generally you find something similar, copy paste, and hack on it until you get the result you are expecting.

stillinbeta, 2022-12-04:
That's sort of what I figured, I'll take a look. Thanks for the
quick response!

stillinbeta, 2022-12-05:
I did a bit more digging, and I think I figured this out, from section 22.4.12 of the full manual. This particular count register can either be 32 bits of counting, or it can be 31 bits of counting and one bit that is "a read-only copy of the UIF bit of the TIMx_ISR register." I did come up with a stanza that removes that:

```yaml
+TIM2:
+  _delete:
+    _registers:
+    - "CNT_ALTERNATE5"
+  CNT:
+    _delete:
+      - CNT_H
+    _modify:
+      CNT_L:
+        name: CNT
+        description: Counter value
+        bitWidth: 32
+
```

but this also crops up in several other places, so I'm just going to find a way to manage this difference in HAL. Closing this for now.

closed 2022-12-05 by stillinbeta

burrbull, 2022-12-05:
One of possible solutions for this is to use methods for access to any register https://github.com/rust-embedded/svd2rust/pull/692

The second is to make 1 of the registers main with direct access and other secondary with method access. But this is more hard to program.