stm32-rs/stm32-rs #810: Request STM32H5 support!

hydra, 2023-03-18:
The STM32 H5 is a new CortexM33 MCU from ST Micros.  ST released all the datasheets, reference manuals, updated cubemx a few days ago.

https://www.st.com/en/microcontrollers-microprocessors/stm32h5-series.html

https://www.st.com/en/microcontrollers-microprocessors/stm32h562rg.html

We need a PAC for it ASAP please!

I can't find the SVD files on ST's website yet, but using Eclipse CMSIS Packs plugin I was able to find a set of SVD's from Keil.  Attached.

[Keil-STM32H5-SVD-v1.0.0.zip](https://github.com/stm32-rs/stm32-rs/files/11009680/Keil-STM32H5-SVD-v1.0.0.zip)

David-OConnor, 2023-03-21:
Initial error from `svd2rust`:

```
[ERROR svd2rust] Error parsing SVD XML file

    Caused by:
        0: In device `STM32H562`
        1: In peripheral `ADC1`
        2: In register `ADC_CFGR`
        3: Parsing field `EXTSEL0` at 854:15
        4: `EnumeratedValue error: Value 2 out of range 0..2
 ```

Relevant section of SVD:

```
  <field>
                <name>EXTSEL0</name>
                <description>External trigger selection for regular group
These bits select the external event used to trigger the start of conversion of a regular group:
...
Note: The software is allowed to write these bits only when ADSTART = 0 (which ensures that no regular conversion is ongoing).</description>
                <bitOffset>5</bitOffset>
                <bitWidth>1</bitWidth>
                <access>read-write</access>
                <enumeratedValues>
                  <enumeratedValue>
                    <name>B_0x0</name>
                    <description>adc_ext_trg0</description>
                    <value>0x0</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>B_0x1</name>
                    <description>adc_ext_trg1</description>
                    <value>0x1</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>B_0x2</name>
                    <description>adc_ext_trg2</description>
                    <value>0x2</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>B_0x3</name>
                    <description>adc_ext_trg3</description>
                    <value>0x3</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>B_0x4</name>
                    <description>adc_ext_trg4</description>
                    <value>0x4</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>B_0x5</name>
                    <description>adc_ext_trg5</description>
                    <value>0x5</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>B_0x6</name>
                    <description>adc_ext_trg6</description>
                    <value>0x6</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>B_0x7</name>
                    <description>adc_ext_trg7</description>
                    <value>0x7</value>
                  </enumeratedValue>
                  <enumeratedValue>
                    <name>B_0x1F</name>
                    <description>adc_ext_trg31</description>
                    <value>0x1F</value>
                  </enumeratedValue>
                </enumeratedValues>
              </field>
```

hydra, 2023-03-21:
> <bitWidth>1</bitWidth>

The width is wrong here right?

[image]

[image]

ST's own documentation is confused too!

My suggestion is to combine the fields into one 5 bit field, called `EXTSEL`, using the values specified in `EXTSEL1` and then delete `EXTSEL0`, `EXTSEL1`, etc.

closed 2023-03-22 by bors[bot]

hydra, 2023-03-22:
Fantastic timely work @David-OConnor and @adamgreig much appreciated, I'm trying the nightlies out now.

https://github.com/stm32-rs/stm32-rs-nightlies/tree/master/stm32h5

🥳

Dirbaio, 2023-03-27:
`stm32-metapac` v2.0.0 has H5 support.

WIP HAL support is here https://github.com/embassy-rs/embassy/pull/1300

commit 78e5071d (2023-03-22): Merge #811

811: Added H5 r=adamgreig a=David-OConnor

Fixes #810
PR #811 (2023-03-22): Added H5
Fixes #810