stm32-rs/stm32-rs #824: How to use svd file in VS Code or STM32 Cube IDE

KamilDuljas, 2023-03-30:
Hi, I would like to use patched svd, but I can not because in VS-Code (Cortex Debug Extension) return error and in STM32 Cube IDE in debug mode I don't see pheripherials:

[image]

Is any manner to fix that?

KamilDuljas, 2023-03-30:
I found some information about similar problem here:
https://github.com/Marus/cortex-debug/issues/513

burrbull, 2023-03-30:
I'd suggest you to use [probe-rs-debugger](https://marketplace.visualstudio.com/items?itemName=probe-rs.probe-rs-debugger)
But I don't know how good it works out of Rust world.

KamilDuljas, 2023-03-30:
Perhaps patched files are not compatible with CMSIS schema?

burrbull, 2023-03-30:
They are. We check them with xmllint against https://github.com/stm32-rs/stm32-rs/blob/master/svd/cmsis-svd.xsd

Although derivedFrom link resolution is not a trivial task, so ST avoid to use them.

burrbull, 2023-03-30:
As alternate solution you could try to expand derived registers with
```
svdtools convert --expand <IN_PATH> <OUT_PATH>
```
before pass SVD to cortex-debug

KamilDuljas, 2023-03-31:
It's better, now It works but output file contains mistakes:
[image]

Enumerate fields do not contain numbers and description (not shown on figure) is set as "H0" for all fields.

It looks like bug in parser. Output file:
```
<register>
          <name>HR0</name>
          <displayName>HR0</displayName>
          <description>digest registers</description>
          <addressOffset>0xC</addressOffset>
          <size>0x20</size>
          <access>read-only</access>
          <resetValue>0x00000000</resetValue>
          <fields>
            <field>
              <name>H</name>
              <description>H0</description>
              <bitOffset>0</bitOffset>
              <bitWidth>32</bitWidth>
            </field>
          </fields>
        </register>
        <register>
          <name>HR1</name>
          <displayName>HR0</displayName>
          <description>digest registers</description>
          <addressOffset>0x10</addressOffset>
          <size>0x20</size>
          <access>read-only</access>
          <resetValue>0x00000000</resetValue>
          <fields>
            <field>
              <name>H</name>
              <description>H0</description>
              <bitOffset>0</bitOffset>
              <bitWidth>32</bitWidth>
            </field>
          </fields>
        </register>
``` 

I tried to modify input file but with no result.  In my opinion, displayName and field Name should have variable syntax to enumerate value
```
<register>
          <dim>5</dim>
          <dimIncrement>0x4</dimIncrement>
          <dimIndex>0-4</dimIndex>
          <name>HR%s</name>
          <displayName>HR0</displayName>
          <description>digest registers</description>
          <addressOffset>0xC</addressOffset>
          <size>0x20</size>
          <access>read-only</access>
          <resetValue>0x00000000</resetValue>
          <fields>
            <field>
              <name>H</name>
              <description>H0</description>
              <bitOffset>0</bitOffset>
              <bitWidth>32</bitWidth>
            </field>
          </fields>
        </register>
```

KamilDuljas, 2023-03-31:
Perhaps is it bug in svdtools during the patch operation?

burrbull, 2023-03-31:
I don't understand what the issue and what you mean under `variable syntax`

KamilDuljas, 2023-03-31:
It means that : 
 <name>HR%s</name> has "%s" to expand (0,1,2....) ,  but 
<displayName>HR0</displayName> and
<name>H</name>
<description>H0</description>
not.
It should looks like:
 <name>HR%s</name>
...
<displayName>HR%s</displayName> 
...
<name>H%s</name>
...
<description>H%s</description>

KamilDuljas, 2023-03-31:
I see two issues.
1. SVD should contains more fields able to enumerate like above.
2. Valid handling this case through svdtools :)

KamilDuljas, 2023-03-31:
Moreover, You said that files are compatible with CMSIS schema but CMSIS tools returned errors after scans. For example svdconv tool from https://github.com/ARM-software/devtools return over 1k+ errors for patched files and 0 errors for vendor files. I know that this repo is kind of fork but It should align with widthly community that using stm32, both Rust and the Others.

burrbull, 2023-03-31:
Saying about `displayName` I could agree. We just never worry about this tag as it is not part CMSIS specification and we've never used it for our goals.

Saying about `H` name of fields they were renamed consciously to collect `HRx` registers in array. See https://github.com/stm32-rs/stm32-rs/blob/master/devices/common_patches/hash/hash_v1.yaml

KamilDuljas, 2023-03-31:
About enhancment:
The user in above linked issue said:
https://github.com/Marus/cortex-debug/issues/513#issue-1034561233

>This SVD contains a enumeratedValues field that ST's SVD does not have.
> thought that the cause of the error was probably that enumeratedValues deriveFrom was a forward >reference, so I changed the order of the SVD description so that the definition of enumeratedValues comes >first, and now it can be loaded.
.....
>I couldn't find a spec on whether forward references are allowed, so I proposed this as an issue, but this may >not be allowed.

Perhaps the changing the order of the definition of enumeratedValues is good point to be align with CMSIS checking tools? It could using patched SVD with widthly group, various extensions related to Cortex debug and official stuff from ST :)

burrbull, 2023-03-31:
About enumeratedValues derivedFrom. I've remembered that historically we are supporting 1 unspecified case - link name (dis_en_enum) without field name (clk) if both derived and linked enums are in same register.
Accordingly to CMSIS there must be `clk.dis_en_enum`: https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_registers.html#elem_enumeratedValues

 Try to change `derivedFrom="ENUM"` with `derivedFrom="FIELD.ENUM"`. Is this solves issue with `Cortex-Debug`.

Saying about `svdconv` checking open different issue for this, please.

KamilDuljas, 2023-04-04:
I think that we could close the issue because other issue has been opened and is a real problem with svd file. When we will be met requirements about the svd file it will be possible to use in VSCode and others IDE.

https://github.com/stm32-rs/stm32-rs/issues/825