stm32-rs/stm32-rs #825: Patched SVD files do not meet the requirements through svdconv tool

KamilDuljas, 2023-03-31:
Example:
svdconv.exe stm32f7x6.svd
return

Found 0 Error(s) and 1 Warning(s).

svdconv.exe stm32f7x6.svd.patched
return
Found 1060 Error(s) and 292 Warning(s).

The patches schould be compatible with CMSIS tools and Vendors architecture manners.

KamilDuljas, 2023-03-31:
In relation to the comment:
https://github.com/Open-CMSIS-Pack/devtools/issues/814#issuecomment-1491972090

forward reference is not supported and this restriction will be added to CMSIS documentation

burrbull, 2023-03-31:
I've made fast fix. Could you try [stm32f7x6.svd.patched.new.zip](https://github.com/stm32-rs/stm32-rs/files/11123643/stm32f7x6.svd.patched.new.zip) ?

KamilDuljas, 2023-03-31:
Still the same. The solution is guaranting the ordering derivedFrom entry against to reference to type as I shown in 
https://github.com/Open-CMSIS-Pack/devtools/issues/814
For example: First declare USART1 and then use derivedFrom="USART1". The svdFile still use forward reference

burrbull, 2023-03-31:
where can I load svdconv for linux?

highly prefered by direct link for ci

KamilDuljas, 2023-03-31:
Compile source only
https://github.com/Open-CMSIS-Pack/devtools

burrbull, 2023-03-31:
it is awful

KamilDuljas, 2023-03-31:
It's binary for Windows:
https://drive.google.com/drive/folders/1g4zcyAUWLmCwDw-jHGXfO2i8zeHHyhHo?usp=share_link

burrbull, 2023-03-31:
I've added field name to derivedFrom which was the real issue. Then I reordered fields to place derived first, but svdconv still not happy:
[image]
```

*** ERROR M206: Z:/home/burrbull/tests/stm32f7x6.patched.supernew.svd (Line 9422) 
  DerivedFrom not found: 'PUPDR0.PUPDR0'

*** INFO M211: Z:/home/burrbull/tests/stm32f7x6.patched.supernew.svd (Line 9422) 
  Ignoring DerivedFrom  (see previous message)

*** INFO M211: Z:/home/burrbull/tests/stm32f7x6.patched.supernew.svd (Line 9422) 
  Ignoring EnumeratedValues  (see previous message)

*** ERROR M206: Z:/home/burrbull/tests/stm32f7x6.patched.supernew.svd (Line 9429) 
  DerivedFrom not found: 'PUPDR0.PUPDR0'
```
[stm32f7x6.patched.supernew.svd.zip](https://github.com/stm32-rs/stm32-rs/files/11127341/stm32f7x6.patched.supernew.svd.zip)

KamilDuljas, 2023-04-01:
Hmm, You need read deeper CMSIS schema :p 
```
            <field derivedFrom="PUPDR0">
              <name>PUPDR1</name>
              <description>Port x configuration bits (y =               0..15)</description>
              <bitOffset>2</bitOffset>
              <bitWidth>2</bitWidth>
            </field>
```
Above example works fine. Look also on USART. It uses "derviedFrom" as attribute of pheripherial node:

```
 <peripheral derivedFrom="USART1">
      <name>USART6</name>
      <baseAddress>0x40011400</baseAddress>
      <interrupt>
        <name>USART6</name>
        <description>USART6 global interrupt</description>
        <value>71</value>
      </interrupt>
    </peripheral>

KamilDuljas, 2023-04-01:
```
<xs:complexType name="fieldType">
    <xs:sequence>
      <xs:group    ref="dimElementGroup" minOccurs="0"/>
      <!-- name specifies a field's name. The System Viewer and the device header file will
           use the name of the field as identifier -->
      <xs:element name="name" type="dimableIdentifierType"/>
      <!-- description contains reference manual level information about the function and
           options of a field -->
      <xs:element name="description" type="stringType" minOccurs="0"/>
      <!-- alternative specifications of the bit position of the field within the register -->
      <xs:choice minOccurs="1" maxOccurs="1">
        <!-- bit field described by lsb followed by msb tag -->
        <xs:group ref="bitRangeLsbMsbStyle"/>
        <!-- bit field described by bit offset relative to Bit0 + bit width of field -->
        <xs:group ref="bitRangeOffsetWidthStyle"/>
        <!-- bit field described by [<msb>:<lsb>] -->
        <xs:element name="bitRange" type="bitRangeType"/>
      </xs:choice>
      <!-- access describes the predefined permissions for the field. -->
      <xs:element name="access" type="accessType" minOccurs="0"/>
      <!-- predefined description of write side effects -->
      <xs:element name="modifiedWriteValues" type="modifiedWriteValuesType" minOccurs="0"/>
      <!-- writeContstraint specifies the subrange of allowed values -->
      <xs:element name="writeConstraint" type="writeConstraintType" minOccurs="0"/>
      <!-- readAction specifies the read side effects. -->
      <xs:element name="readAction" type="readActionType" minOccurs="0"/>
      <!-- enumeratedValues derivedFrom=<identifierType> -->
      <xs:element name="enumeratedValues" type="enumerationType" minOccurs="0" maxOccurs="2">
      </xs:element>
    </xs:sequence>
    <xs:attribute name="derivedFrom" type="dimableIdentifierType" use="optional"/>
  </xs:complexType>
```

As above part of CMSIS schema I see that derivedFrom is attribute of field, not enumeratedValues

burrbull, 2023-04-01:
> Hmm, You need read deeper CMSIS schema :p
>
> ```
>             <field derivedFrom="PUPDR0">
>               <name>PUPDR1</name>
>               <description>Port x configuration bits (y = [...]

Deriving of whole field is independent question. You are right saying that this is better in this particular case.
But this doesn't give answer why svdconv can't find enumeratedValue. It doesn't say "whole field should be derived here", it says "DerivedFrom not found"

KamilDuljas, 2023-04-01:
As I said above. Not found because enumerated value has not a attribute in schema

burrbull, 2023-04-01:
enumeratedValue**s** has https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_registers.html#elem_enumeratedValues

Oh. @adamgreig could you comment this?
https://github.com/stm32-rs/stm32-rs/blob/371fc811387c760f3b92e030aa7c68d3250b3ba8/svd/cmsis-svd.xsd#L23-L30

KamilDuljas, 2023-04-01:
Ah, ok. My mistake.

Perhaps you should extend naming such as
    peripheral + register + field?

KamilDuljas, 2023-04-01:
Ok I found some solution:
The name of enumeratedValues must be different than field name:
```
<field>
              <name>PUPDR0</name>
              <description>Port x configuration bits (y =               0..15)</description>
              <bitOffset>0</bitOffset>
              <bitWidth>2</bitWidth>
              <enumeratedValues>
                <name>PUPDR0A</name>
                <usage>read-write</usage>
```

and derivedFrom must be full "namespace": 
```
<enumeratedValues derivedFrom="GPIOD.PUPDR.PUPDR0.PUPDR0A"/>

Otherwise It does not work. Maybe It is bug in svdconv or the next idocumentation issue and implicit assuming that the names must be different :)

KamilDuljas, 2023-04-01:
I think that open next issue on svdconv

burrbull, 2023-04-01:
Thanks.

> The name of enumeratedValues must be different than field name:

This should be easy to fix

> and derivedFrom must be full "namespace":

but this could be problematic, especially during patching. Еhe shorter the path, the less conflicts.

KamilDuljas, 2023-04-01:
I know, full namespace sounds like bug... Or feature
:D

burrbull, 2023-04-01:
> I know, full namespace sounds like bug... Or feature :D

It is feature if we have ready unbugged SVD.
But if we need to change it, or even more copy parts from one to another then it's not funny anymore.

KamilDuljas, 2023-04-01:
To tracking:
https://github.com/Open-CMSIS-Pack/devtools/issues/815

KamilDuljas, 2023-04-04:
@burrbull please refer to the proposed solution in https://github.com/Open-CMSIS-Pack/devtools/issues/815

thorstendb-ARM, 2023-04-06:
Hi, some comments:
- SVDConv is available as compiled for Windows, Linux, Mac, and as source distribution. See https://github.com/Open-CMSIS-Pack/devtools
- regarding derivedFrom:
-- if field name and enumeratedValues container name are the same, it falls into a "this" trap, which fails afterwards, because the types do not match. See SvdItem.cpp, FindChildFromItem().
- regarding forward references:
-- currently SVDConv does not support forward references. This is because derivedFrom can be executed to an object or it's children that have already been derived from another object, a dimed object, or both. SVDConv model stores and links original data and references to other objects (derivedFrom and dim), but currently forward references (which means, that derivedFrom is executed after reading the whole model) are not supported.

burrbull, 2023-10-15:
@KamilDuljas with https://github.com/rust-embedded/svdtools/pull/173 and https://github.com/stm32-rs/stm32-rs/pull/320 I've got 
[stm32f7x6.svd.patched.zip](https://github.com/stm32-rs/stm32-rs/files/12909500/stm32f7x6.svd.patched.zip)

svdconv shows:
```
Found 5 Error(s) and 295 Warning(s).
```