stm32-rs/stm32-rs #851: `_cluster` block build

eZioPan, 2023-06-11:
I'm trying to doc `OTG_FS_DEVICE` peripheral for STM32F4. and I encounted a problem.

There is only 4 End Point for `OTG_FS_DEVICE` in F429 ([RM0090]), thus only 1-3 for `DIEP%s` and `DOEP%s`, but in [fs_device_cluster.yaml], it clusters 1-5. As long as I don't `_add` or `_delete` fields of the real exist registers, `svdtools` and `svd2rust` can generate correct libs (only 1-3 generated). But if this registers have been modified, then `svdtools` refuse patch svd files.

Here is [the log from CI]. The [line of yaml file] and [another line of yaml file] I think may related to this.

Since [fs_device_cluster.yaml] is included in `fs_v1` and `fs_v2`, I'm not sure should I split this file into 2 (or even more?) versions to include in different devices (which may affect many devices), or `svdtools` may need to do some modification, that it will check whether a member of cluster real exist before generate patches?

[RM0090]: https://www.st.com/resource/en/reference_manual/rm0090-stm32f405415-stm32f407417-stm32f427437-and-stm32f429439-advanced-armbased-32bit-mcus-stmicroelectronics.pdf

[fs_device_cluster.yaml]: https://github.com/stm32-rs/stm32-rs/blob/8fb5ba9bad963cb44282c3b7bfce1c0e2b3d533a/devices/common_patches/usb_otg/fs_device_cluster.yaml

[the log from CI]: https://github.com/stm32-rs/stm32-rs/actions/runs/5234629817/jobs/9450918700?pr=849#step:7:168

[line of yaml file]: https://github.com/stm32-rs/stm32-rs/pull/849/files#diff-ef7a6f03b66e74dbf8df3da290de8bc9b292a2e8e3cb7fa295d18ab311b18eb5R338

[another line of yaml file]: https://github.com/stm32-rs/stm32-rs/pull/849/files#diff-ef7a6f03b66e74dbf8df3da290de8bc9b292a2e8e3cb7fa295d18ab311b18eb5R206

burrbull, 2023-06-12:
> but in [fs_device_cluster.yaml](https://github.com/stm32-rs/stm32-rs/blob/8fb5ba9bad963cb44282c3b7bfce1c0e2b3d533a/devices/common_patches/usb_otg/fs_device_cluster.yaml), it clusters 1-5

It is not true.
`DIEPCTL[1-5]` pattern means it takes all registers matches this pattern.
So if there is no `DIEPCTL4` and `DIEPCTL5` it just ignore them. It fails only if no matches.

You need first **delete** `DIEPCTL4` if it is not really present in chip **before** collecting registers in cluster.

eZioPan, 2023-06-12:
> You need first **delete** `DIEPCTL4` if it is not really present in chip **before** collecting registers in cluster.

I have try to `_delete` `DIEPCTL4` and `DIEPCTL4`, but `_cluster` still report error.

Then I comment out entire **_delete** and **fs_device_cluster.yaml**. Libs are generated, and no `DIEP*[4-5]` and `DOEP*[4-5]` exist. There are only end point 0-3.

I might think originally there are no end point 4 and end point 5 exist for these devices, but `_cluster` some how create them.

eZioPan, 2023-06-12:
I just comment out include line from device yaml, and it works. thanks!