stm32-rs/stm32-rs #859: `svd/extract.sh` must be run from `svd` directory, instructions do not say this

cbiffle, 2023-06-27:
Instructions say

> Run svd/extract.sh to extract the zips into svd (ignored in git).

Great!

```
$ svd/extract.sh
unzip:  cannot find or open vendor/*.zip, vendor/*.zip.zip or vendor/*.zip.ZIP.

No zipfiles found.
mv: cannot stat 'stm32h7a3x.svd': No such file or directory
mv: cannot stat 'stm32h7b0x.svd': No such file or directory
mv: cannot stat 'stm32h7b3x.svd': No such file or directory
mv: cannot stat 'stm32h73x.svd': No such file or directory
mv: cannot stat 'stm32h742x.svd': No such file or directory
.....
```

The script assumes it's being run from `svd`. It'd be straightforward to fix this (e.g. add a `cd $(dirname $0)`), or the instructions could be updated to point out the need for a two-step `cd svd && ./extract.sh`.

cbiffle, 2023-06-27:
Ah, I note that _some_ of the instruction lists spell this out: "Generating Devices" does, "Adding new devices" and "Updating existing devices" do not. FWIW.

adamgreig, 2023-06-27:
Thanks for the report. Now that `make` runs `extract` automatically it's not needed at all in "Generating devices", but I've updated adding/updating to run `make extract` instead which will `cd` as appropriate.

commit f594d1b6 (2023-07-04): Update README instructions around extracting SVDs; closes #859
PR #861 (2023-07-04): Update README instructions around extracting SVDs
Closes  #859
closed 2023-07-04 by adamgreig