stm32-rs/stm32-rs #862: stm32h5 release

astapleton, 2023-07-22:
It appears that there is a crate for stm32h5, but no releases have been made (crate is at version 0.0.0). What is the status of support for STM32H5 and is there going to be a crate release in the near future? Does it need contributions?

adamgreig, 2023-07-22:
This will get released with the 0.16 release of all the PACs; I hope to do that very soon as it's quite overdue, but I need to work out which PRs to include before the release. In the meantime you can test it at https://github.com/stm32-rs/stm32-rs-nightlies/tree/master/stm32h5 and report any issues.

astapleton, 2024-02-05:
If anyone else is interested in this, the 0.15.1 crate has now been published. It's derived entirely from the ST SVD file, and is probably usable, but doesn't include any patches reflected on master. To get those, use the nightly repo.

closed 2024-02-05 by astapleton

adamgreig, 2024-02-05:
Just to clarify, the 0.15.1 release does include all patches for h5 as of the time it was released (i.e. it's current master, not master from 2022 when the other crates' 0.15.1 was released).