stm32-rs/stm32-rs #863: smt32f7 v0.10 and above break simple LED blink firmware

Qix-, 2023-07-23:
The following code works on `stm32f7 = { version = "0.9", features = [ "stm32f7x6", "rt" ] }` but not on `stm32f7 = { version = "0.10", features = [ "stm32f7x6", "rt" ] }` (0.9 vs 0.10):

```rust
#![no_main]
#![no_std]
use core::panic::PanicInfo;
use stm32f7::stm32f7x6::Peripherals;

#[no_mangle]
fn main() -> ! {
	let per = Peripherals::take().unwrap();
	per.RCC.ahb1enr.write(|w| w.gpioben().bit(true));
	per.GPIOB.moder.write(|w| w.moder7().bits(0b01));

	let mut on: bool = true;
	loop {
		per.GPIOB.odr.modify(|_, w| w.odr7().bit(on));
		on = !on;

		for _ in 0..100000 {
			unsafe {
				core::arch::asm!("NOP");
			}
		}
	}
}

#[panic_handler]
fn panic(_panic: &PanicInfo<'_>) -> ! {
	loop {}
}
```

My linker script `memory.x` has this:

```ld
MEMORY
{
  FLASH (rx) : ORIGIN = 0x08000000, LENGTH = 1024K
  RAM (xrw)  : ORIGIN = 0x20000000, LENGTH = 320K
}

_stack_start = ORIGIN(RAM) + LENGTH(RAM);
```

Was there a change around then that requires different setup?

Qix-, 2023-07-23:
Ah okay, starting in `0.10` you have to switch to `thumbv7em-none-eabihf` instead of just `thumbv7em-none-eabi`. I wish the docs had some more usage about this, I was just trying random things based on a bunch of outdated guides I've found. This was the fix.

Sorry for the noise!

closed 2023-07-23 by Qix-

adamgreig, 2023-07-23:
No worries! Though I'm quite surprised at the fix - either target should work fine, the only difference is whether the FPU is enabled at startup and then FPU registers used when passing floats to functions. I wonder if anything else changed when moving targets, maybe even just forcing the recompilation was enough. Hopefully you can be using 0.15 now, though, as 0.10 is pretty old.

Qix-, 2023-07-23:
Yeah once 0.10 started working, switching to 0.15.1 also worked fine. And no worries, I know this stuff gets pretty hairy as time move forward :)

Thanks for the quick response either way!