stm32-rs/stm32-rs #867: I need step by step to blink led in my stm32f469 disco kit Using Rust

neuberfran, 2023-08-08:
Hi,

I stay trying this: https://stackoverflow.com/questions/75882337/how-can-i-run-rust-code-on-a-stm32f415-target
in my stm32f469 disco kit unsuccessfully.

stm32f469 disk kit has BSP? Or will I have to start with cargo init?

https://community.st.com/t5/stm32-mcu-products/issue-to-adapt-code-to-make-a-simple-blink-led-in-stm32f469/m-p/579588/thread-id/220050

newAM, 2023-08-08:
There is no step by step guide for your specific need, but here are some resources that can help you get started.

First, I recommend learning some of the basics of the rust language before diving into embedded rust, rustlings is great for this: https://github.com/rust-lang/rustlings

Then the rust-embedded book is a great starting point for learning the embedded-specific parts of rust: https://docs.rust-embedded.org/book/unsorted/

Then for your specific problem of blinking an LED on the stm32f469 discovery kit check the examples in the stm32f4xx-hal: https://github.com/stm32-rs/stm32f4xx-hal/blob/f4aea7607dec1157d0b694ea5998753a9ca69a68/examples/blinky.rs#L19

burrbull, 2023-08-08:
> stm32f469 disco kit has BSP? Or will I have to start with cargo init?

I've added "Fast start" section. https://github.com/stm32-rs/stm32f4xx-hal#fast-start

Saying about BSP. You could create your own and suggest community to merge it.
But IMHO BSPs don't give big advantage to spend forces to support them.

neuberfran, 2023-08-08:
@newAM @burrbull Tks. I used cargo generate and the template for **stm32f469nih6** and put the code from the link below and I was successful in compiling
https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/delay-syst-blinky.rs

As the photo, rtic I put as fake. @burrbull  would like your opinion on this

I also got the error: :3333: Connection timed out. In the command below:
(gdb) target remote :3333

And my last question is about the example code from the link above: delay-syst-blinky.rs, in which the author uses pa5 and (as shown in the photos below) I imagine they are Arduino pins (because the 4 LED pins built into the board didn't work before)

[image]
[image]
[image]

burrbull, 2023-08-08:
Yeah. This board needs board support layer. At least for pins:
[image]

burrbull, 2023-08-08:
> I also got the error: :3333: Connection timed out. In the command below:
> (gdb) target remote :3333

What tool/command do you use for flashing?

neuberfran, 2023-08-08:
@burrbull I'm relying on this readme below: https://gist.github.com/neuberfran/7019716c8d1d746b410a1ca70cb937c3

In vscode after getting success in cargo build I run:
1) sudo openocd -f interface/stlink-v2-1.cfg -f target/stm32f4x.cfg
2) arm-none-eabi-gdb -q -ex "target remote :3333" target/thumbv7em-none-eabihf/debug/rust-stm32f4-disco-blinky

Or I rely on the link below, generating the *.hex
1) arm-none-eabi-objcopy -O ihex -R .eeprom target/thumbv7em-none-eabihf/debug/rust-stm32f4-disco-blinky rust-stm32f4-disco-blinky.hex 
2) sudo openocd -f interface/stlink-v2-1.cfg -f target/stm32f4x.cfg -c init -c halt -c "program rust-stm32f4-disco-blinky.hex" -c exit

https://stackoverflow.com/questions/75882337/how-can-i-run-rust-code-on-a-stm32f415-target

burrbull, 2023-08-08:
I propose to forget `openocd` if no other option.
`probe-rs` tools are much easier to use:
https://probe.rs/docs/tools/cargo-flash/

neuberfran, 2023-08-08:
> Yeah. This board needs board support layer. At least for pins: http://www.st.com/st-web-ui/static/active/en/resource/technical/document/user_manual/DM00218846.pdf
> 
> [image]
> 
> [image]
Could you explain this better?

burrbull, 2023-08-09:
In "Pin name" column there is Arduino compatible name on your board.

The real names of pins are in "MCU Pin" column. There is no PA5 pin.
According to "Table 8" there should be 4 pins connected to leds on the board. For green led you need to change `gpioa -> gpiog`, `pa5 -> pg6`.
[image]

Or connect your own led to any pin from "Table 5"

neuberfran, 2023-08-09:
To resolve this issue I referred to this video of mine on the channel: https://www.youtube.com/watch?v=4rV624d6xv8

In the st community forum they told me (as shown in the photo below) that pa1 is equal to d3 in the arduino pinout of the stm32f469 disco kit.

I also started using cargo flash --release --chip stm32f469 ih6 as per the link below:
https://probe.rs/docs/tools/cargo-flash/

https://www.youtube.com/watch?v=TKCirpm_SSw

Note: 1) I looked at the datasheet and I still don't know why pa1 is equal to d3
2) I change Cargo.toml in [dependencies.stm32f4xx-hal]
version = "0.11.1" to "0.13.1"
[image]

closed 2023-08-09 by neuberfran