stm32-rs/stm32-rs #869: `stm32-rs` `v0.15.1` cannot compile with `stm32f412` feature enabled

eZioPan, 2023-08-15:
rustc says:
```
error: symbol `DEVICE_PERIPHERALS` is already defined
X:\blah\blah\stm32f4-0.15.1\src\stm32f412\mod.rs:2307:1
static mut DEVICE_PERIPHERALS: bool = false;
```

I can confirm this bug is not exist in master branch.

So, any plan to release a new version?

adamgreig, 2023-08-17:
Are you definitely not linking in two different versions of `stm32f4`? You could check with `cargo tree -d`. I just checked a demo project with `stm32f4` 0.15.1 and `stm32f412` feature enabled and it built without issues.

eZioPan, 2023-08-18:
Oops, I can confirm `stm32f4` with `stm32f412` enabled build without any error.

My project is a `[workspace]`, old crates depend on `stm32f4` with `stm32f411` enabled, and my newest crate depened on `stm32f4` with `stm32f12` enabled. And the error comes out.

I guess using different `feature` sets of same dependency in diffrent crates in the same `workspace` case this linking error?

Uhm, maybe a missing feature in rust build system?

eZioPan, 2023-08-18:
ok, I'm not going to fight with linker, I will just move every crates to depend on stm32f412.

thanks!

closed 2023-08-18 by eZioPan