stm32-rs/stm32-rs #883: STM32F103. The peripherals that are missing in the microcontroller are specified

AndreySmirnov81, 2023-09-21:
Peripherals that are not in the microcontroller and therefore should not be specified:
- Ethernet_MMC, Ethernet_MAC, Ethernet_PTP, Ethernet_DMA, 
- TIM9, TIM10, TIM11, TIM12, TIM13, TIM14,
- OTG_FS_DEVICE, OTG_FS_GLOBAL, OTG_FS_HOST, OTG_FS_PWRCLK.

burrbull, 2023-09-21:
**Ethernet (ETH): media access control (MAC) with**
DMA controller
Low-density devices are STM32F101xx, STM32F102xx and STM32F103xx
microcontrollers where the Flash memory density ranges between 16 and 32 Kbytes.
Medium-density devices are STM32F101xx, STM32F102xx and STM32F103xx
microcontrollers where the Flash memory density ranges between 64 and 128 Kbytes.
High-density devices are STM32F101xx and STM32F103xx microcontrollers where the
Flash memory density ranges between 256 and 512 Kbytes.
XL-density devices are STM32F101xx and STM32F103xx microcontrollers where the
Flash memory density ranges between 768 Kbytes and 1 Mbyte.
Connectivity line devices are STM32F105xx and STM32F107xx microcontrollers.
This section applies only to STM32F107xx connectivity line devices.

**USB on-the-go full-speed (OTG_FS)**
Low-density devices are STM32F101xx, STM32F102xx and STM32F103xx
microcontrollers where the Flash memory density ranges between 16 and 32 Kbytes.
Medium-density devices are STM32F101xx, STM32F102xx and STM32F103xx
microcontrollers where the Flash memory density ranges between 64 and 128 Kbytes.
High-density devices are STM32F101xx and STM32F103xx microcontrollers where the
Flash memory density ranges between 256 and 512 Kbytes.
XL-density devices are STM32F101xx and STM32F103xx microcontrollers where the
Flash memory density ranges between 768 Kbytes and 1 Mbyte.
Connectivity line devices are STM32F105xx and STM32F107xx microcontrollers.
This section applies only to STM32F105xx and STM32F107xx connectivity line devices.

AndreySmirnov81, 2023-09-22:
Brief excerpt:

**Ethernet (ETH): media access control (MAC) with DMA controller**
applies only to STM32F107xx connectivity line devices.

**USB on-the-go full-speed (OTG_FS)**
applies only to STM32F105xx and STM32F107xx connectivity line devices.

And there is no mention of STM32f103 here.