stm32-rs/stm32-rs #913: STM32G491 missing FDCAN2 peripheral

liamkinne, 2023-10-26:
It appears to be missing in the svd from ST (i.e. it is present in the G484 svd, but not the G491)

Would submit a PR myself but I can't find any examples on how one would add entire peripheral with a patch rather than the individual registers.

burrbull, 2023-10-26:
You can copy it from another SVD. Something like:
```
_copy:
  FDCAN2:
    from: ../svd/stm32g491.svd:FDCAN2
```

liamkinne, 2023-10-26:
@burrbull thanks! Will give that a go.

liamkinne, 2023-10-26:
Running into this error.

```
[2023-10-26T20:46:34Z ERROR svdtools::cli] by svdtools (0.3.4)
    
    Caused by:
        0: Processing device `STM32G491xx`
        1: According to `FDCAN,FDCAN?`
        2: Processing peripheral `FDCAN2`
        3: According to `RXGFC`
        4: Could not find `FDCAN2:RXGFC. Present registers: .`
make: *** [svd/stm32g491.svd.patched] Error 1
```

Not sure what to do about this since the RXGFC register is definitely in the SVD.

burrbull, 2023-10-27:
In g484 it is derived from FDCAN.

```yaml
_derive:
  FDCAN2:
    _from: FDCAN
    baseAddress: 0x40006800

FDCAN2:
  add:
    _interrupts:
      FDCAN2_intr0:
        description: FDCAN2_intr0
        value: 86
      FDCAN2_intr1:
        description: FDCAN2_intr1
        value: 87
```

PR #915 (2023-11-06): Add FDCAN2 peripheral to G491
Resolves #913
closed 2023-11-06 by adamgreig