stm32-rs/stm32-rs #971: STM32H7 update to new .svd (v1.9) to fix wrong address for HSEM_CnIER, HSEM_CnICR, HSEM_CnISR, HSEM_CnMISR

taunusflieger, 2024-03-25:
I'm working with a STM32H755. I recognised that the .svd files used (STM32H755_CM4 and STM32H755_CM7) are from May 2019. They are incorrect regarding the addresses of the HSEM_CnIER, HSEM_CnICR, HSEM_CnISR and HSEM_CnMISR registers. They contain for both files (CM4 and CM7) the same addresses for the above registers which is wrong according to RM0399 chapter 11.4.3. The correct naming and addressing (as reflected in the current version v1.9 of the [STM32H7_svd](https://www.st.com/resource/en/svd/stm32h7-svd.zip)) would be (HSEM_C1IER, HSEM_C2IER), (HSEM_C1ICR,HSEM_C2ICR) ,(HSEM_C1ISR, HSEM_C2ISR) and (HSEM_C1MISR HSEM_C2MISR). HSEM_C1xx are the registers for core 1 and HSEM_C2xx are the registers for core 2.

I'm happy to support here, but I don't know the process of upgrading to a new version of the svd file. Any hints are welcome.

burrbull, 2024-03-26:
You could try to restore #951 PR.
Although it looks like some patches in it use incorrect syntax.

taunusflieger, 2024-03-27:
Looking at this PR and the complexity involved when updating to a new version of the svd I tent to approach this differently - creating a patch for the current svd to get the addresses fixed

burrbull, 2025-02-09:
Source archives are updated. Closing.
Reopen if issues still present.

closed 2025-02-09 by burrbull