stm32-rs/stm32-rs #976: [STM32H7A3] Undefined references

cracksalad, 2024-04-09:
This is my very first attempt to program my STM32s using Rust. So the following might totally be my fault:

When building basically the snippet from the README.md...

```rust
use stm32h7::stm32h7b3::Peripherals;

fn main() {
    let mut peripherals = Peripherals::take().unwrap();
}
```

<details>
<summary>...the following error shows up when building on Windows (with v0.15.1 as well as nightly):</summary>
```
STM32H7A3Test.4wxjn0cmolgiuilv.rcgu.o : error LNK2019: unresolved external symbol __primask_r referenced in function _ZN8cortex_m8register7primask4read17h31510c12cde55e95E
STM32H7A3Test.5ajxmlxw3sywah7d.rcgu.o : error LNK2019: unresolved external symbol __cpsie referenced in function _ZN8cortex_m9interrupt6enable17hc5c3f77bc5f1dcd0E
STM32H7A3Test.5ajxmlxw3sywah7d.rcgu.o : error LNK2019: unresolved external symbol __cpsid referenced in function _ZN8cortex_m9interrupt7disable17hc5498504712ca0efE
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol WWDG
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol PVD_PVM
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol RTC_TAMP_STAMP_CSS_LSE
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol RTC_WKUP
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol FLASH
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol RCC
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol EXTI0
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol EXTI1
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol EXTI2
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol EXTI3
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol EXTI4
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol DMA1_STR0
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol DMA1_STR1
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol DMA1_STR2
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol DMA1_STR3
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol DMA1_STR4
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol DMA1_STR5
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol DMA1_STR6
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol ADC1_2
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol FDCAN1_IT0
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol FDCAN2_IT0
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol FDCAN1_IT1
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol FDCAN2_IT1
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol EXTI9_5
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol TIM1_BRK
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol TIM1_UP
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol TIM1_TRG_COM
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol TIM1_CC
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol TIM2
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol TIM3
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol TIM4
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol I2C1_EV
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol I2C1_ER
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol I2C2_EV
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol I2C2_ER
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol SPI1
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol SPI2
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol USART1
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol USART2
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol USART3
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol EXTI15_10
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol RTC_ALARM
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol TIM8_BRK_TIM12
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol TIM8_UP_TIM13
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol TIM8_TRG_COM_TIM14
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol TIM8_CC
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol DMA1_STR7
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol FMC
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol SDMMC1
libstm32h7-46862da39392481e.rlib(stm32h7-46862da39392481e.stm32h7.1ebe771ea1f23660-cgu.0.rcgu.o) : error LNK2001: unresolved external symbol TIM5
[... cut ...]
/usr/bin/ld: .../target/debug/deps/STM32H7A3Test-9c274c5e43b34279.4k8ctir7bhgi1kwv.rcgu.o: in function `critical_section::release':
~/.cargo/registry/src/index.crates.io-6f17d22bba15001f/critical-section-1.1.2/src/lib.rs:200: undefined reference to `_critical_section_1_0_release'
collect2: error: ld returned 1 exit status
```
</details>

All the errors are about undefined references/external symbols. I am using rustc v1.77.1 and want to program a STM32H7A3ZIT6Q. Am I doing something wrong here? Any hint is welcome!

burrbull, 2024-04-09:
have you enabled "rt" and "critical-section" features?

burrbull, 2025-02-09:
Closing. Reopen if issue is still present on nightly builds.

closed 2025-02-09 by burrbull