stm32-rs/stm32-rs #997: Release soon?

liamkinne, 2024-08-01:
There are a lot of unreleased changes and it has been over two years since the last release. Is there a plan to release new crate versions soon?

If there's anything blocking this I am happy to help out.

burrbull, 2024-08-02:
Only @adamgreig can answer the question.

Some time ago I've made another crate as temporary solution: https://crates.io/crates/stm32f4-staging

burrbull, 2024-08-02:
> If there's anything blocking this I am happy to help out.

Usually Adam beautify CHANGELOG.md before release adding PR links and grouping by family.

What status of #951 ? Do you plan to revive it?

liamkinne, 2024-08-02:
@burrbull I've given up on that PR because there were too many differences between the old and new svd files to reconcile easily and there isn't much to gain from it since the new svd still has a lot of errors in it.

kairoswater-jason, 2024-10-28:
The STM32G0 crate has not been updated in 3 years, but the SVD was updated to 1.5V in this repo 3 months ago (and since then ST released V1.6 of the stm32g0 svd file with even more bugfixes)

The 3 year old version of the STM32G0 crate is missing registers from my target MCU (that were added in the 1.5V SVD).

burrbull, 2025-04-29:
@usbalbin We plan new release. Do you have any suggestions/fast fixes?

cc @adamgreig

usbalbin, 2025-04-29:
Yay!

I do not have anything right now. I (or @Wratox) will hopefully start working with the H5 in a couple of weeks. Since it is quite new there might be some new findings there. But no, nothing concrete right now or even soonish.