stm32-rs/stm32-rs #1027: HRTIM - Naming master timer vs normal timer

usbalbin, 2024-11-16:
@burrbull

All of [HRTIM_MASTER's registers](https://docs.rs/stm32g4-staging/0.18.0/stm32g4_staging/stm32g474/hrtim_master/index.html) are prefixed with an `m` while some of the [other timers registers](https://docs.rs/stm32g4-staging/0.18.0/stm32g4_staging/stm32g474/hrtim_tima/index.html) are prefixed with `tim`

Some examples 

| HRTIM_MASTER | HRTIM_TIMA |
|-------------------------|--------------------|
|    **m**cmp1r   | cmp1r             |
|    **m**dier       |**tim**dier    |

Would it make sense to strip those prefixes to make the naming more consistent?

burrbull, 2024-11-16:
Yes. Makes sense. I'll look what can I do.

burrbull, 2024-11-16:
Should we rename `MPER => PERR`, `MREP => REPR`?

burrbull, 2024-11-16:
Try this branch: https://github.com/stm32-rs/stm32-rs/tree/hrtim-master

usbalbin, 2024-11-16:
That is really nice!

I also noticed that the HRTIM_TIMERx's capture register 2 `cpt2r` seems to contain the bitfield `cpt1` in your branch. Would it make sense if both `cpt1r` and `cpt2r` just contained the bitfield `cpt`?

burrbull, 2024-11-16:
> I also noticed that the HRTIM_TIMERx's capture register 2 `cpt2r` seems to contain the bitfield `cpt1` in your branch. Would it make sense if both `cpt1r` and `cpt2r` just contained the bitfield `cpt`?

Should be fixed in last commit.

burrbull, 2024-11-16:
Open PR, pls.

burrbull, 2024-11-17:
Check and open PR from https://github.com/stm32-rs/stm32-rs/tree/hrtim-common

burrbull, 2024-11-20:
https://github.com/stm32-rs/stm32-rs/tree/hrtim-rstr

closed 2024-11-24 by usbalbin

usbalbin, 2024-11-25:
@burrbull would you mind creating a new release of [stm32g4-staging](https://crates.io/crates/stm32g4-staging) with the new hrtim changes?

burrbull, 2024-11-26:
done