stm32-rs/stm32-rs #1040: Where are the enums

usbalbin, 2024-12-05:
@burrbull or anyone

When trying to update hals to use the latest pacs one common thing I stuble upon is missing enums(or the things that used to have a `_A`-suffix. For example

```
cannot find type `QSPISEL` in module `rcc::d1ccipr`
[...]
help: a type alias with a similar name exists: `QSPISEL_R`
```

Looking at FMCSEL_R i see `pub use FMCSEL_R as QSPISEL_R;`
and `FMCSEL_R` in turn: `pub type FMCSEL_R = crate::FieldReader<FMCSEL>;`.

So I take it that I should use `FMCSEL` instead of `QSPISEL`?

Why are there nice type aliases for `FOO_R` types but not for `FOO` (earlier `FOO_A`)? Is this intended behaviour?

burrbull, 2024-12-05:
No `_A` suffixes more.

> Why are there nice type aliases for FOO_R types but not for FOO (earlier FOO_A)? Is this intended behaviour?

In short, yes. They created a lot of conflicts and implementation difficulties.
So now svd2rust or:
1. default. It uses `enumeratedValues` name for enums where possible and field names for nameless enumeratedValues`, minimum type aliases.
2. with `--field_names_for_enums` option it uses field names for enums with type aliases everywhere.

> So I take it that I should use FMCSEL instead of QSPISEL?

1. Yes (create user type alias).
2. Split https://github.com/stm32-rs/stm32-rs/blob/2a8b02236f4176418ac94cbaa412b841fc4d6e92/peripherals/rcc/rcc_v3_h7_ccip.yaml#L33-L37 rule on separate copies to prevent reexporting. (better in this particular case)
3. Add `_name` which good describes common enum. (better in other cases).
https://github.com/stm32-rs/stm32-rs/pull/1033#issuecomment-2490181907

usbalbin, 2024-12-05:
Ok. Thanks