stm32-rs/stm32-rs #1122: Add H523 and H533

usbalbin, 2025-01-20:
From what I can tell SVD's for H523 and H533 are now available https://www.st.com/resource/en/svd/stm32h5-svd.zip

Wratox, 2025-01-23:
For `stm32h5 = { package = "stm32h5-staging", version = "0.17.0" }` this `rcc.pll1cfgr().modify(|_, w| w.pll1m());` worked for stm32h503, stm32h562, stm32h563 and stm32h573 but for `stm32h5 = { git = "https://github.com/stm32-rs/stm32-rs-nightlies" }` `pll1m()` seems to have been renamed to `divm1()` for all of the above except stm32h503 and stm32h523.

usbalbin, 2025-01-23:
[RM0481](https://www.st.com/resource/en/reference_manual/rm0481-stm32h52333xx-stm32h56263xx-and-stm32h573xx-armbased-32bit-mcus-stmicroelectronics.pdf) uses the name `PLL1M` (see 11.8.7).

@burrbull should this be changed to pll1m? or should it be divm1?

burrbull, 2025-01-23:
`pll1m`

usbalbin, 2025-01-24:
divmX vs pllXm was fixed in #1128 and H533 support was added in #1129.

closed 2025-01-24 by usbalbin

Wratox, 2025-01-27:
For stm32h523 and stm32h533 APB3ENR has `VREFBUFEN()` while the other have `VREFEN()`, see https://stm32-rs.github.io/stm32-rs/STM32H523.html#RCC:APB3ENR and https://stm32-rs.github.io/stm32-rs/STM32H533.html#RCC:APB3ENR

Also found that APBx number in description doesn't match number of the register for https://stm32-rs.github.io/stm32-rs/STM32H563.html#RCC:APB3LPENR
Same with https://stm32-rs.github.io/stm32-rs/STM32H563.html#RCC:APB3ENR