stm32-rs/stm32-rs #1130: How to build the last official release v0.15.1? What versions of `svdtools` and `svd2rust` were used for that version?

pietgeursen, 2025-01-23:
## Context

I'm using https://github.com/David-OConnor/stm32-hal which uses stm32-rs v0.15.1.

If I use the latest nightlies from this repo they have obviously changed a lot since v0.15.1 and generate an overwhelming number of errors.

As an alternative I'd like to patch the most important changes I need on top of v0.15.1.

## Issue

I can't get v0.15.1 to build. This is without any patched changes, just trying to build at that commit. I've tried various combinations of versions of `svdtools` and `svd2rust` but all with various different errors.

Anybody solved this problem before? Am I doing something silly?

adamgreig, 2025-01-24:
0.15.1 used svd2rust 0.24.1 and svdtools 0.2.3, according to the [CI at the time](https://github.com/stm32-rs/stm32-rs/blob/v0.15.1/.github/workflows/ci.yaml).

burrbull, 2025-02-09:
I think it is answered. So closing.

closed 2025-02-09 by burrbull