stm32-rs/stm32-rs #1154: How to handle field value as flags instead of enum ?

vinchatl, 2025-02-02:
I'm trying to document the `OPTCR` and `OPTCR1` registers of the FLASH peripheral for STM32F4 devices, but I don't know how to handle the `nWRP` field.

The FLASH memory of STM32F4 devices is split into several sectors, 12 or 24 sectors depending on device and memory configuration. The `nWRP` field is 12 or 24 bits long (to match the number of sectors of the FLASH memory), with each bit specifying the write protection of the corresponding sector.

This means the value stored in `nWRP` shouldn't be treated as an enumeration but rather as a flags, with each flag corresponding to the write protection status of a sector.

Here's an extract of ST documentation:

[image]

[image]

I tried (without success) to create a patch for `OPTCR` and `OPTCR1` registers using `_split` which looks like this

```yaml
OPTCR:
  nWRP:
    _split: 
      nWRPx:
        name: nWRP%s
        description: Value of the write-protection option bytes after reset for sector %s
```

But the `svdtools` command fails with the following message

```bash
svdtools patch devices/stm32f405.yaml
[2025-02-02T10:50:12Z ERROR svdtools::cli] by svdtools (0.4.3)
    
    Caused by:
        0: Processing device `STM32F405`
        1: According to `FLASH`
        2: Processing peripheral `FLASH`
        3: According to `OPTCR`
        4: Processing register `OPTCR`
        5: Processing field matched to `nWRP`
        6: Adding read-write enumeratedValues
        7: `EnumeratedValues error: EnumeratedValues is empty
        8: EnumeratedValues is empty
make: *** [Makefile:50: svd/stm32f405.svd.patched] Error 1
```

Is this the correct way to do this ? Is it even possible ?

burrbull, 2025-02-02:
> This means the value stored in `nWRP` shouldn't be treated as an enumeration but rather as a flags.

Agree. This should be array of fields.
There 2 way.
First is (not recommended as breaks structure of patches):
1. _delete _nWRP
2. Add `WRP%s` array
3. Add enums

Second is
1. Use `_spilt` to get `WPR0`, `WPR1`, etc.
2. Add enums to `WPR?`
3. Collect  `WPR?` with  `_array`

Ideally all 3 steps in different files in 1st in `patches`, 2nd in `fields` and 3rd in `collect`

burrbull, 2025-02-02:
> Does it look good to you ?

Yes. The only point is what `n` prefix means? Is this a "number" or something `Flash` related?

vinchatl, 2025-02-02:
Thanks :)
I'll check manuals for all STM32F4 devices and update impacted ones.

The `n` prefix inverts the write protection status, so `0` means the sector is protected while `1` means it isn't protected (so basically, `1` means 'sector is writable').
It it wasn't there, `1` would mean protected.

burrbull, 2025-02-02:
This field is present on all F2, F4 and F7 devices.

vinchatl, 2025-02-02:
Then I'll check F2 and F7 docs as well.

There are at least 2 variants, i.e. variant A with OPTCR for F405 and version B with OPTCR+OPTCR1 for F427. There might be more considering the number of devices.

Do you prefer a single PR with everything or several PR divided by family (or some other criteria) ?

burrbull, 2025-02-02:
I'd prefer all-in-one if possible.

burrbull, 2025-02-02:
Life hack how to control your changes.

1. Run `cargo make yamls` before applying patches to generate YAML-like SVDs.
2. Copy `yaml` folder somewhere.
3. Patch
4. Rerun `cargo make yamls`.
5. Use `meld` or another difftool to see changes.
6. Repeat steps 3-5.

vinchatl, 2025-02-02:
Well well, after an initial search over F2-F4-F7 manuals, there are 7 differents variants for `OPTCR` register, 2 variants for `OPTCR1` register, and even an `OPTCR2` register for some F7 devices…
It'll be a bit longer than I thought.

I saw H5 and H7 also have `option bytes` registers but there's a lot of them and their structure is radically different, so I won't target them.

closed 2025-02-03 by burrbull