stm32-rs/stm32-rs #1187: Are Core Registers not included on purpose?

Jonarw, 2025-04-02:
I use patched SVD files from stm32-rs for embedded STM32 C/C++ debugging using Cortex-Debug, as they seem to be much higher quality than the 'originals' provided by ST. Specifically I am currently working on STM32G0B1. I noticed that the Core Registers, like NVIC and SCB, are not included in the SVDs. Glancing through this repo, this seems to be the case for most MCUs, not just G0. However I did find e.g. `devices/fields/nvic/nvic_v1.yaml`, indicating that there is / used to be at least some support for these registers.

My main question is: Are Core Registers out of the scope of this repo and therefore purposely not included? Or is it mainly a matter of nobody having gotten around to adding/patching/fixing them?

burrbull, 2025-04-02:
> However I did find e.g. `devices/fields/nvic/nvic_v1.yaml`, indicating that there is / used to be at least some support for these registers.

We are using original SVD provided by ST and then patch them. There were some core peripherals in old versions of SVD but were deleted after update.
Possibly it is because they are not ST specific and common across all Cortex-M.

We could add some extra patching level (disabled by default) which adds missing core peripherals for debug purposes. cc @adamgreig 

In fact these peripherals are already present in cortex-m directory. All that is needed is to modify build system.

Jonarw, 2025-04-03:
Ah I see, makes sense!
I tried if I can hack something together myself, but I ran into some issues.

Specifically, if I run `svdtools armv6m.yaml` in `cortex_m/` I get this error:
```
[2025-04-03T08:06:34Z ERROR svdtools::cli] by svdtools (0.4.5)
    
    Caused by:
        0: Processing device `ARMv6-M`
        1: Adding peripheral `CPUID`
        2: `Build error: `usage` must be initialized
        3: `usage` must be initialized
```

I tried adding the `usage` field to `addressBlock` in `cpuid_v6.yaml`, but still receive the same error. I guess I don't know enough about the expected format of the yaml to understand why this is the case... Any pointers would be welcome.

burrbull, 2025-04-03:
No.

You should add
```yaml
_include:
  - ../cortex_m/armv6m.yaml
 ```
in stm32g0b1.yaml

And then run `svdtools patch stm32g0b1.yaml`

Something like this.

Jonarw, 2025-04-03:
Thanks for your help.
That is exactly what I tried first, but I get the same error message:
```
svdtools patch stm32g0b1.yaml
[2025-04-03T08:33:55Z ERROR svdtools::cli] by svdtools (0.4.5)
    
    Caused by:
        0: Processing device `STM32G0B1`
        1: Adding peripheral `CPUID`
        2: `Build error: `usage` must be initialized
        3: `usage` must be initialized

```

I only tried `svdtools patch armv6m.yaml` to try and debug the issue.

burrbull, 2025-04-03:
I remembered. There is issue https://github.com/rust-embedded/svdtools/issues/282 I forgot about.

Try to fully remove addressBlock. Maybe it is enough for you.

Jonarw, 2025-04-03:
Yes, removing `addressBlock` from all relevant files in cortex_m/peripherals did the trick! The resulting SVD is good enough for debugging purposes. Thanks for your support!

Of course it would be very nice if there was a way to get complete patched SVDs without tinkering, but I fully understand that this is not the main purpose of this repo.

closed 2025-04-03 by Jonarw