stm32-rs/stm32-rs #1210: pyproject for the scripts?

jaskij, 2025-05-01:
I'm working on adding STM32N6 parts (PR coming soon!), and wanted to run the script `matchperipherals.py`. Unfortunately, there is no project description or the like.

This creates two difficulties:
- figuring out what the dependencies are
- manual creation and management of a venv (`pip install` outside of virtual environments is deprecated and strongly discouraged, for good reasons)

To me this creates quite a high barrier for entry. Having a proper Python project would allow me to do something like `$(poetry env activate)` or `hatch shell` to easily create and activate a virtual environment and run the scripts.

burrbull, 2025-05-01:
matchperipherals is outdated and useless for now as it does not support inperipheral _includes

jaskij, 2025-05-01:
It is still recommended in `CONTRIBUTING.md` though